meerk40t/meerk40t #2086: [0.8.10] Duplicated image shows wrong size in tree label

Sophist-UK, 2023-09-03:
[image]
I loaded a design, set the DPI to 500 and then duplicated the group the image was in. Whilst both images show the correct size in properties, the label is incorrect.

jpirnay, 2023-10-11:
I can't reproduce the issue - is it still in place? And what does the label show - the original dpi / dimensions?

Sophist-UK, 2023-10-11:
I am sure that it is still happening in 0.8.10 - why would it stop happening?

jpirnay, 2023-10-19:
It's fixed in the 0.9.2 release but did not make it in time for the 0.8 bugfix release. Will be arriving there with the next 0.8 bugfix release cycle.

closed 2024-01-17 by jpirnay