meerk40t/meerk40t #2091: Rubber Stamp Mode

tatarize, 2023-09-09:
It may be reasonable to create an effect that primarily runs within image operation. This would do the primary effects needed for stamps:

1. Image thresholding. All image goes to either black or white.
2. Image flip. We must flip horizontally so the stamp comes out correctly.
3. Image invert. The black sections are marked to not contain inked sections.
4. Image ramping. We create ramping from a certain distance away from the black sections. Largely this would be like a blur with white sections mixing with black, but no black mixing with the white sections. This allows a gentle slope of stamp rather than a steep cliff.

In theory this should be done as an extension of effect image operations, running sort of as a parallel to image-wizard as such.

Sophist-UK, 2023-09-09:
I would certainly be interested in this.

I would also be interested in a similar mode for carving other materials whereby the layers are generated using increasing (or decreasing) shades of grey. So each layer would be black or white based on a grey threshold, and the threshold would increase on higher layers e.g. on a 4 layer carve, 1st layer would be light grey or darker, 2nd would be mid-grey or darker, 3rd would be dark grey/black and the 4th would be black.

Sophist-UK, 2023-09-09:
P.S. This would need to apply to Raster ops as well as image ops. (For example, if I want to create a stamp with lettering on it.

The UI would need to be carefully considered. The scene would need to give an accurate representation of what will be burned.

I am not sure it would be parallel to image-wizard - partly because it would need to be available to Raster Ops too. For image Ops, I am not sure whether it would be a replacement for Image-Wizard or run in parallel with it.

tatarize, 2023-09-09:
In theory we could also, nominally render this data, if that was needed. Basically if you have just an image or set of images we process them according to these rules. If you have vector objects in there, just render them.

The idea generally for how to have `effect hatch` and `op hatch` sort of agree with each other is to allow engrave to have a `effect hatch` node within it. We'd then be able to deprecate `op hatch` if we merely allowed hatch effect. Currently if you type `effect-hatch` it adds to the elements an `effect hatch` node. Effect nodes basically rely on the 0.9.x change that all element nodes contain an `as_geometry()` or `as_image()` function which provides a geomstr object or an image object. The `effect hatch` node is an element which accepts as operands closed shapes and will fill them with it's hatchy goodness. This happens in the regular scene and does not require a special operation to do this.

This solution is a lot better than the `op hatch` solution which only compiled the hatch at cut planning time. The best solution I've come up with for bridging the gap is to see that `op engrave` can contain an `effect hatch` and when it does the `references` get added to the effect. And when this is put in the cut planner. This ends up being the same thing as an engrave containing hatch-effect containing operands. It just means that rather than containing a reference to the hatch-effect within the `elements branch` it actually contains a real hatch-effect node. And the references rather than being contained by the engrave are contained by the effect.

---

This sort of dual use of an effect, means that you wouldn't necessarily need these effects to be within elements branch. They could be there, but they could also be contained wholly by an operation.

This has some clear implications for how `RasterWizard` should work. Currently it's a bit confusing and feels sort of hidden for being a pretty great feature. Rather we'd wrap a much more basic `elem image` inside an `effect rasterwizard` sort of thing. This would accept an image and return a properly raster-preprocessed image. This would also make it a bit more obvious in the tree what exactly is going on (at least hopefully). But, this also means we could have an `elem image` inside an `effect stamp` which would do all the regular stamp stuff rather than raster wizard stuff. We'd just have the node work a bit like a function. Taking an image and returning an image when you call the effects `as_image()` function. 

In theory, within this scheme we'd actually be able to subsume all `op raster` nodes too. Since, as it turns out, `op raster` is really just an `op image` containing what amounts to a `effect render`. Since all we're doing for rasters that we don't do for images is taking all the vectors objects and rendering them.

Obviously this could get confusing if we dove in headfirst and went purely with the modified paradigm. But, there are probably some more subtle ways to basically do this behinds the scenes and generally just sort of let you add effects to operations and get these sorts of things. Like, for example, the operands of `hatch-effect` get confusing if you're allowed to see them in the tree so they are mostly hidden, we could likewise do the same sort of thing with ops.

---

So in theory, I could make an image operation that contained a `stamp-effect` sort of thing. And it could appear as an `op image` with a `effect stamp` inside it. Or I could maybe flag the image op as containing a pre-processor for images (as a purely generic thing) and maybe rename the node `Stamp Operation` kind of thing. I'm not 100% sure which sort of setup would be best. -- I do expect to be able to remove `op hatch` either as an `op engrave` containing a `hatch-effect` or as an `op engrave` flagged with a `effect hatch` preprocessor. 

Perhaps the latter, later giving way to the former, so it could appear more seamless and just divided a bit more correctly in code. And the letting people see the man behind the curtain.

Also, there are a few minor things like how you'd make an engrave with an effect, because dragging and dropping a `hatch-effect` into one seems like a bad idea since that already drags a reference of that for geometry purposes.

tatarize, 2023-09-09:
> If you would like me to experiment with a PoC version of MK with these functions, I would be happy to do so. We have some rubber stamp material unused and a fair amount of scrap wood & acrylic.

The simplest proof of concept of this is `Z-Divide` which should work. And subdivide a zdepth image into sub-images. But, you're absolutely right that this scheme could totally be used to do `3D images` like that. Where we slice the image on the fly or according to whatever specifications we need at the time.