meerk40t/meerk40t #2103: Switch translations protocols from wxPython to python.

tatarize, 2023-09-13:
Update meerk40t to use python specific gettext methods rather than wxPython. 

```python
el = gettext.translation('meerk40t', localedir='locale', languages=['it'])
el.install()
_ = el.gettext
```

For example, this creates a `_` function that uses the gettext from the local with the name meerk40t. This example does `it` for Italian. 

This would allow the translations to be flagged and operated within the kernel, and to permit the translations of kernel level messages. This also means it would permit translated info even in the `main.py` CLI options. This would make some of the kludgy methods for passing around the gui-level translations unneeded. This could also permit some simplified channels that are intended to be routinely translated. Since you could name the set the name of the channel to `_` and commands like `_()` would actually both flag this as text needing translations and serve as the function that calls the channel (with some translation functions).

tiger12506, 2023-09-13:
I would exercise caution translating internal items, as an example, translating channel names might make it difficult to communicate the command necessary to watch/open/save a particular channel.
I think it's a great idea for help though, such as the output of ? and ?? (just not any command names, etc.)

tatarize, 2023-09-13:
Commands can't ever really translate.

Though, we could maybe better divide up the files being used into sub-files. So that we can show users what the most important things to translate is. Do the gui stuff, maybe don't do the way cryptic print shutdown stuff. And maybe have them as different folders or whatever. Still rather speculative but it's a fundamental part of the program and translations were built in from the start... and making the gettext stuff less wxPython dependent would seem like a good move.

closed 2024-10-04 by jpirnay