meerk40t/meerk40t #2111: [0.9.2 Pre Beta 2] This was release as a full release and is being offered to 0.8 users!!!!!!

Sophist-UK, 2023-09-19:
I assume that this was not intended.

tatarize, 2023-09-19:
Ooof.

tatarize, 2023-09-19:
The discord now has crash logs create discord entries for the crashes in the crash-reports forum. So the crash logs basically get pretty fast feedback. I was likely going to maybe nuke that version or not even offer the 0.9.2 beta line anymore which is probably better. At least until 0.9.1 is finished up.

tatarize, 2023-09-19:
I also have Homeassistant have google home mini read off the title of any github issue raised.

Sophist-UK, 2023-09-19:
Apologies then for the early enter on the other issue I just raised which had a mostly blank title and no description. 🤣 

I would keep the 0.9.2 pre-beta - just edit the tag and make it pre-release.

tatarize, 2023-09-19:
Yeah, just that the 0.9.2 stuff is a bit useful but a bigger change than I would want in 0.9.1 (moves to views away from matrices there).

Sophist-UK, 2023-09-19:
Exactly!

closed 2023-09-27 by tatarize