meerk40t/meerk40t #2112: [0.8.10

Sophist-UK, 2023-09-19:
This is a:

*   [] Bug
*   [] Suggestion for an enhancement

## SUMMARY

## DETAILS

## ERROR MESSAGE
```
<!-- Please copy and paste any error messages from e.g. a crash report file. -->
```

## YOUR ENVIRONMENT

*   MeerK40t version:
*   OS: 
*   Running from: <|-- e.g. Executable, package (PIP3/Pypi), downloaded source, git managed source

### SCREENSHOTS

### Zipped SVG File(s)

Sophist-UK, 2023-09-19:
I took a quick look at the code and my supposition that the code changes in #776 were not included in 0.8 was **incorrect** - so this is not the cause.

tatarize, 2023-09-20:
I don't see the problem. It looks like it burns the inners first when it's checked.

Sophist-UK, 2023-09-20:
It's not burning the tiny circles in sequence - those circles which are over the empty image corners get skipped until the image has been burned and then it goes back and fills in those tiny circles that were missed before going elsewhere off the bottom of the screenshot to do other inner burns.

So it looks like it is treating the small circles as burn inner first closed cut paths (when they are Engrave ops not Cut ops) and not burning them until the inner image has been burned.

But the code seems to select Cut ops for outer paths ... so I am not sure what is happening.

tatarize, 2023-09-20:
Yeah, but it's not burning the outer-part before any of the current parts. So the internal stuff is correct. It's burning inner first. Just because the inner part is seemingly less optimal doesn't make it a bug, maybe a defect. But, it's obeying the set constraints.

Sophist-UK, 2023-09-20:
The algorithm is not working as it should and the results are sub-optimal. Whether you call it a bug or a defect or an error or unoptimised, either way it is not working as it should.

As for two-opt, I believe that I have a new and much much more efficient algorithm for 2-opt if you are interested.

tatarize, 2023-09-20:
You'd be hard pressed to beat 2-opt. Especially the current geomstr numpy optimized version of it. Though sometimes you can do greedy then 2-opt and get better than just straight 2-opt at least for time.

Sophist-UK, 2023-09-20:
Incredible - you know without even hearing how my more efficient algorithm works that it is not as good as "geomstr numpy".

tatarize, 2023-09-21:
> Incredible - you know without even hearing how my more efficient algorithm works that it is not as good as "geomstr numpy".

It's pretty hard to beat geomstr numpy version of two-opt. I'm pretty sure k-opt is probably superior but, you can't really do better than nearly instant fantastic nearly optimal optimizations. You would be hard-pressed to beat it. I didn't say it was impossible, but as far as good approximations of NP TSP it's really really hard to beat two-opt. And for speed it's nearly impossible to beat my implementation there.

tatarize, 2023-09-21:
Also, I'm fairly sure you outlined it (or something like it) in #1949

Sophist-UK, 2023-09-21:
Your right, I did outline it in #1949 but you have provided no evidence or analysis to show that the two-opt algorithm is going to be faster. I would agree that any algorithm that is capable of being run in numpy has a good chance of being faster than one that can't, but there is no evidence to show whether my algorithm could be run in numpy.

So in reality we don't know whether:

a. Which would be faster as non-numpy versions of both algorithms; or
b. Whether my algorithm can be numpied; or
c. If it can which numpy algorithm would be faster; or
d. If my algorithm can't be numpified, then which would be faster regardless.

Sophist-UK, 2023-09-21:
That said, this is a law of diminishing returns. Greedy is a fantastic improvement over file order and saves 90%+ of the travel moves - lets call this 90%+ optimised. An almost instant 2-opt (even on huge complex files) will improve them to (say) 99%.

The odd sub-optimal travel move after than is going to make no appreciable difference whatsoever after that.

There are other optimisations that can be done which might give bigger benefits that tweaking the last microgram out of travel optimisations:
* The non-overlapping raster optimisations can save a heap of time when non-overlapping rasters are separated in the raster direction by significant distances. 
* Similarly breaking a raster image (Image or Raster op) into segments where there are big white gaps (think of braking a picture frame into two L-shaped rasters as an example) can also save a heap of time.
* Optimising the direction of rastering so that the travel is to the nearest corner, so that you decide at run time whether to do T2B or B2T, and in which direction you raster the first row would reduce travel time a small amount before the raster.

closed 2024-10-04 by jpirnay