meerk40t/meerk40t #2139: [Bug]: keyboard issue

ludovicjardel, 2023-10-04:
### Summary Description

No possibility to enter a value in the Speed (mm/s) field on Linux Mint (latest) 
Only "6 key" is possible. Other keys have no effect.

### Additional Details

I use a laptop. I tried to plug a usb keyboard with a numpad and it works fine.
I also tried ctrl-v and it's ok

### MeerK40t Version

08.1002

### MeerK40t Type

Git

### Your Operating System

Linux

tatarize, 2023-10-05:
In `keymap` when you hit the key what does it say that key is?

ludovicjardel, 2023-10-05:
> In `keymap` when you hit the key what does it say that key is?

it's the same key (eg : if I press "2" key "2" is highlighted)
even if I select another keyboard (I have a french azerty keyboard)

tatarize, 2023-10-05:
Ah, that clarifies things...

https://stackoverflow.com/questions/56566858/wx-keyevent-and-non-qwerty-keyboards

So you hit the `é` key and get a 2 to come up or some such thing?

ludovicjardel, 2023-10-05:
I must use shift é to get a 2.That's how it works on all applications (except this field of meerk40t)

ludovicjardel, 2023-10-05:
I tried on another PC and got the same issue. No problem on windows

tatarize, 2023-10-05:
The post there says I can switch to the char event and it will translate that rather than going for the hard-coded key.

ludovicjardel, 2023-10-05:
Should I understand that you can fix it ? It should be great as we have 2 K40 in the fablab I manage and the users don't understand why they should use ctrl-v to select the speed :-)

tatarize, 2023-10-05:
Maybe. I think the reason I went for KEY_DOWN rather than CHAR_HOOK before was because there was some need to for commands that worked while the key was held down that stopped when the key was lifted back up again. This is something that CHAR_HOOK maybe unable to do. 

The easiest thing may be to rebind the commands. If that seems reasonable enough. You can in keymap rebind them if you wish. So that you can have multiple keystrokes that do the same thing. I may be able to process any `+alias` and `-alias` commands with keydown but reserve the charhook for everything else.

tatarize, 2023-10-05:
I think we understand it now, so it's not that strange but there's a lot of differences between key_down/key_up and charhook. We'll look for a more nuanced solution but, in the mean time, you can rebind those keys to anything you want. Especially something with ascii keys below 127.

tatarize, 2023-10-05:
I thought we were more discussing hotkeys rather than number typing into the special text entry fields. We've updated the discussion and it may be an alternative functionality thing to the text boxes that we can more easily rectify.

tatarize, 2023-10-05:
I szitched ,y keyboqrd to azerty qnd I cqn type nu,bers into the boxes zithout qny issue:

tatarize, 2023-10-05:
I switch my keyboard to azerty and I can type numbers into the boxes without any issue. Though I'm using windows so that's likely not a correct metric.

tatarize, 2023-10-06:
https://github.com/wxWidgets/Phoenix/issues/2464

Is raised, the workaround is live in main and will be in the next beta.

https://github.com/meerk40t/meerk40t/commit/1a6623c249e0534793ee18a0e0d931acaa6e649b

tatarize, 2023-10-06:
The problem is that in Linux the `GetUnicodeKey()` returns the non-shifted key value on keypress there. So it ignores that you held shift for the function we called to figure out whether we should filter those typed letters out... and consequently filter them out.

ludovicjardel, 2023-10-06:
I'm afraid I am totally lost (I am not a developper). But I trust that you will find a solution ;-)
Thanks !

tatarize, 2023-10-06:
Yeah, we found the issue. Raised it with the GUI authors and have added a local workaround.

ludovicjardel, 2023-10-06:
Excellent !!! Thanks a lot