meerk40t/meerk40t #2154: [0.9] Toolbar Look and Feel issues

jpirnay, 2023-10-12:
MeerK40t 0.9 introduced a new ribbon bar addressing a couple of non-resolvable constraints of the one used so far. The look and feel of these versions is different in a lot of details opening the need to address usability and appearance. The original discussion can be found [here](https://github.com/meerk40t/meerk40t/issues/2022) and should be referenced to get the details of the concerns and recommendations.
The original thread had grown in size and addressed issues that have been resolved in the meantime. This new issue is intending to focus on the remaining issues.

- [x] This currently uses a Serif font whereas the whole of the rest of MK uses a Sans Serif font. 
- [ ] Colour background fill too dark
- [ ] No colour background
- [ ] Highlight colors too dark
- [x] Round corners of group boxes _were reduced in size but have been there in  previous versions too_
- [x] Truncated text
- [x] Bottom Margin
- [x] Centre the tab titles in the tabs
- [x] We have lost the design tab. _We have added a toolbar designer in settings where you can add that again if you like_
- [x] Gap between tabs perhaps should be narrower
- [x] Minimum pane width is too narrow
- [x] Vertical-only layout is way too inflexible
- [x] Tooltips do not appear when you mouseover a disabled tool
- [x] Wordlist button has lost its dropdown submenu _see comment above_
- [ ] Icons are fuzzy/illegible. Slightly large or thicker-lined icons would be beneficial _This would require a major overhaul of the icon scheme (and the terms of use of relying on icons8)_
- [ ] Routine to assign space to panels does not make use of additional space vertically / horizontally to reduce the need for the 'additional buttons' indicator

tatarize, 2023-10-13:
The icons issue may be being addressed: #2157

Basically buying svg icons from icons8 and recoding a bit of the icons stuff to go ahead and render vector icons (we do every part of svg rendering already).

[image]

Some of these are vectors (first, 4th, 5th, 6th pane).

tatarize, 2023-10-13:
For a better example (yes I fixed stop)...

[image]

See that the laser icon for `Operations` is converted the vector and the other large icons in there are not converted and still PNG icons there.

tatarize, 2023-10-13:
Also, I tried turning on high res mode for windows:
[image]

When this is tried, keep in mind that we're happy to nuke it and go back to what we had before if it causes too many scaling issues. There's a few different weird things and it may need additional work to scale up the fonts according to some other scaling.

https://github.com/meerk40t/meerk40t/releases/tag/0.9.1002

It looks much more fine-line and crisp on my machine, but it also changed a few things to look weird, and be alien sizes.

Sophist-UK, 2023-10-14:
That does look good (except for the colourless nature of the toolbar.

Sophist-UK, 2023-10-14:
> Some of these are vectors (first, 4th, 5th, 6th pane).

Weirdly the down arrows for Camera and Wordlist have a kink in them.

jpirnay, 2023-10-15:
[image]
This is still WIP, but you get an idea where this is leading to. The overhaul of all the icons is definitely a time-killer...

Sophist-UK, 2023-10-16:
That looks good. As for being a time killer...

I can quite believe that low level UI changes are more time consuming than other stuff. And this is never as sexy as creating new functionality - so it is understandable why it currently doesn't get high priority.

But equally, UI/UX is vitally important - even more important than new functionality IMO.

That said, I am not convinced that adding more UI options is the way to go - it just makes MK even more daunting for newcomers than it already is, rather than simplifying the UI (by making it more task oriented i.e. you get a subset of the UI options depending on the phase / task in your project you are undertaking). Sometimes you need to make users adapt their user habits rather than give them infinite tweaking ability - and I say this as an OCD tweaker that likes the concept of the MK panes even if I hate the difficulties I sometimes have to get the exact UI I want. 

As you add more functionality, the UI inevitably becomes more complex unless you take steps to simplify it by grouping functionality. Inkscape has done this though clearly they are only covering design. There is a toolbar which selects what design task you want to do - and that changes the other toolbar to give you the tools you need. Paint.net does the same thing (though the tool selection is floating).

IMO the existing tabs are the right UI approach, but they are NOT currently task focused. IMO we need to think about the project lifecycle and make the UI match it (i.e. with tabs for each phase). IMO the project lifecycle is this:

1. Design - importing existing SVG / images or drawing them, then scaling and positioning. This probably needs to be split into vector drawing tools and image tools. I would recommend taking inspiration from Illustrator & Inkscape for vector tools, and Photoshop / Paint.net for image tools. Areas for new functionality - fill patterns (both traditional raster fill patters and vector engrave fill patterns), basic image tools, clip paths, full text support (the stuff that currently needs to get Inkscape to fix). The difficulty is knowing which bits to implement and which to leave to specialist tools (like Inkscape or Paint.net / Gimp / Imag-R). Tracing is nice to have, but this is difficult to get right - you might want to see if an existing free tool will partner to allow you to use their algorithms / code (I can recommend one). 

2. Burn prep - defining Operations and their parameters. New functionality possible: Image defaults set on the op and applied to all images in the Op at burn time unless explicit settings were applied to individual images, a much better materials library where you can set defaults for each type of Op for each material.

3. Burn prep - step repeat, word replacement etc. for multiple burns of the same design in one go. The UI in this area is not great.

4. Burning -  what we have now is pretty good (except for the Execute multiple click design and loss of the 0,7 raster optimisations) - but multi-project capability (having several projects loaded at once so you can be burning a couple whilst you prep the next ones, prompting between queued jobs).

0. Configuration - global preferences, devices and their preferences.

+ wizards to help new users get started without it being too daunting.
+ better console help (to make it easier for experts to use the command line)

tatarize, 2023-10-19:
Yeah, the only reason for the ribbonbar, basically drawn from scratch. Is that there were simply thing the other ribbonbar couldn't do (like display horizontal tools). And I thought that the drawing logic wasn't that needed and we really already had hacked through all the other logic to make it do a lot more things.

 I then regretted the giant time suck over the 5x the amount of time I thought it would take.

--

I'm not convinced more options and features is the way to go either. Seems like it sort of needs to be built around workflows. And yeah, some of the proposed GUI stuff here might not be exactly what I would do but, it's clearly the case that currently these sort of workflow things are tied into each other too strongly. Also, from a workflow perspective I don't find myself being omniscient and knowing exactly the settings I'll need before I use them. And really I think burn-prep and burning needs to more generally fit together better.

I often find myself burning something and using that burning to adjust the prep for the next burn or the rest of the current burn. Doing a well defined and known task repeatedly seems to be the least common workflow.

There's clearly often times that the task I'm doing is something a few basic tools can hack out in a few minutes. And that's clearly right and great. But, sometimes when you burn these designs you end up going back to the drawing board and you need to modify the stuff that you already did and you don't want to go all the way back to inkscape for an an adjustment.

I'm not sure making everything modal is right. Where we should expect that that path people take is design, prep, burn. When really it's sort of a connected grid between these so much so that burning and designing are as often done in that order as they are done in the design->burn order.

I think workflow centric is the right way to go, but I'm not sure that flow is always in the same direction. Usually you need to burn things a bit to figure out the correct design, or settings to use.

Though there are certainly a workflow where you have the things set and you just need to do this same thing 100x times. But, also there's a lot of case when you need to do a thing something between 20-30 times but you're not really sure until you see the progress.

I often find myself executing one step at a time, and alternating which step to use. `raster` `raster` `engrave` `raster` ... with the same core settings because I'm busy feeling out the right way to go.

Sophist-UK, 2023-10-21:
> I'm not sure making everything modal is right. 

I am not suggesting that anything should be modal i.e. a window on top and unable to do anything else whilst that window is open.

> Where we should expect that that path people take is design, prep, burn. When really it's sort of a connected grid between these so much so that burning and designing are as often done in that order as they are done in the design->burn order.

I am not suggesting that you can't switch freely between the different phases, just that we need to group functions that are used to achieve one phase separately from functions that are related to a different phase in order to avoid the screen becoming too cluttered and confusing.

> I often find myself executing one step at a time, and alternating which step to use. `raster` `raster` `engrave` `raster` ... with the same core settings because I'm busy feeling out the right way to go.

Yes - whilst I now have a good feel for settings for some sorts of designs and materials, sometimes I need to switch back and forth. BUT if it is a single click to switch between functional groups, this shouldn't be a problem.

tatarize, 2023-11-12:
I'm not sure if this is completed. The vector icons was the last major thing there.

Sophist-UK, 2023-11-12:
See issue I logged earlier today. The ribbon bar is pretty good except that it doesn't scale up far enough.

This can be closed in favour of #2243.

tatarize, 2023-11-13:
Closed in favor of #2243

closed 2023-11-13 by tatarize