meerk40t/meerk40t #2159: [Suggestion] Rate UI control - default 100%

Sophist-UK, 2023-10-13:
The idea here is to have a rate control (in %) which adjusts the actual burn speed accordingly. So if you start a burn and cuts are heavier than needed you can turn up this control to (say) 120% and instead of cuts at 5mm/s, it will do them at 6mm/s instead.

So as MK goes from CutPath to CutPath, varying the speed per the operation that CutPath originated in, it will still vary speed, but modified by the rate control value.

(Under the covers, I guess there would be some sort of Rate command-line command which the GUI would call.)

This might be best located on the Spool window or the Laser pane.

tiger12506, 2023-10-13:
This exists in MeerK40t for GRBL. It is my understanding that the K40 hardware can't really support this properly.

Sophist-UK, 2023-10-13:
Since the k40 m2 nano controller is dumb, MK already has to tell it exactly what speed to use, so MK would only need to apply the factor before telling the controller what speed to use.

The only wrinkle that might need to be considered is **whether or not** to apply the speed change to images and how to change the speed mid-image (which is likely to be more complex).

tiger12506, 2023-10-13:
You'd have to ask Tat for sure, but I think that is the exact issue. The m2nano does not handle speed changes on the fly at all well. He tested it pretty thoroughly, trying to find a way to make it work, IIRC.

Sophist-UK, 2023-10-13:
Speed changes on vector burns shouldn’t be too difficult (even if it means a pause / resume).

I am guessing that speed changes on images will mean stopping the image at the end of a raster row, and effectively starting a new image at the beginning (and position) of the next row – this seems doable, but would need some code and some testing (at ultra-low resolution – when I did this a few months ago for a real burn, I spotted an issue with alternate rows being offset which no one had previously spotted).

tatarize, 2023-10-16:
It's not terrible for on the fly speed changes. I fed in some code to do on-the-fly speed changes at one point it wasn't terrible the biggest issue was that each segment or cut had their own speed and rather than a speed scaling of all future speeds it would quickly reset. So it was good for like one raster cut. And though I couldn't quite figure out why the success rate was not perfect.

You can hop out of the complex speed mode and do `NV<speedcode>SE` and it will update the speed. Though for rasters that would actually invoke a step which is a problem. You can only change speed in rapid mode, and any step out of rapid mode in something like raster-mode could trigger step. Though I think if you reconfirmed the directionality which you could know since it would be stored in current driver state. It would avoid issues and let you pop back in. Even if you couldn't you could do a longer @NSEG002N1SE command which would avoid any raster step since you'd in theory end raster mode for a sec.

tatarize, 2023-10-16:
I checked the code for the `speed` command in lihuiyu is ancient the context doesn't even exist and it dates back to 0.6.x or so.

I'll see about reviving it for a branch or something. Maybe give it a secret command in main.

tatarize, 2023-10-16:
The driver code was still there but the command was ancient. It requires a `lihuiyu` command context (which hadn't existed since 0.6.x) used `speed` which was a bit too general since that's used for a few things that are much less hacky, it called `driver.set()` which didn't call `set_speed()`, probably because it would trigger on the fly mode change. 

So in the bug fixes I fixed the hidden command. It's now called `device_speed` and still should allow some things like `device_speed 90%` and `device_speed -d -5%`.

I don't know if it still works so use with care.

tatarize, 2023-10-16:
It's in the bug fixes for 0.9.2 so it'll move along accordingly.

closed 2024-10-04 by jpirnay