meerk40t/meerk40t #2177: [Bug]: Can not install 9.2 receive message trojan or unwanted software message

ajguerini, 2023-10-19:
### Summary Description

Can not install 9.2, meerk40t.exe, receive message "trojan or unwanted software" message. 
My system
win 11 64 bit
edge explorer

### MeerK40t Version

trying to install 9.2

### MeerK40t Type

Executable

### Your Operating System

Windows

Sophist-UK, 2023-10-19:
It is not a trojan - the reason you get this message is because the project does not have a code signing certificate and because the software used to create the executable from Python is also used by trojan writers.

You can safely accept the risk of opening this application.

closed 2023-10-19 by ajguerini

tatarize, 2023-10-20:
Run from source and it won't do that. The .exe are convenience. 

https://www.youtube.com/watch?v=EaW17iUk0FA

Though that means the source might be less easy.

ajguerini, 2023-10-20:
thank you I did go ahead and run using the exe, once I figured out how to allow my system to run the program. Just started using Meerk40t 9.2 and there have been a couple of crashes and report was sent
Albert

 
 

https://www.youtube.com/watch?v=EaW17iUk0FA

Sophist-UK, 2023-10-20:
> Run from source and it won't do that. The .exe are convenience.

@tatarize I am afraid that whilst this statement may be technically true if we consider that MK could still be used if running from source was the only option, you would not have nearly as many users if this was indeed the only means of running MK. This comment only serves to demonstrate a lack of empathy with mere mortals who simply want to run their laser and who don't have the time or don't have the technical skills to run MK from source.

The EXE files are NOT simply a convenience for most users - users who would never have considered running MK if they had to do all the steps needed to run from source.

I have the technical skills to run from source, and indeed for a while I contributed to the source code which required running from source on my development PC, but for various reasons I choose NOT to run from source on my production PC for both time and packaging reasons.

Leaving aside the time and technical skills to run MK from source, having a stable point in time version of MK in a single file that can be run simply by double-clicking on it is a significant benefit. Having e.g. 0.8.12 and 0.9.2 individually named and being able to easily choose which to run is extremely useful. 

(And before you chip in, yes, I know that you can have multiple source directories and set up shortcuts to run each from source, but that is far more time consuming when there is a new release than downloading an executable and double-clicking it.)

Sophist-UK, 2023-10-20:
> thank you I did go ahead and run using the exe, once I figured out how to allow my system to run the program. Just started using Meerk40t 9.2 and there have been a couple of crashes and report was sent

@ajguerini Albert - sorry to hear that you have been having crashes - MK is normally remarkably stable and crash free. The good news is that @tatarize and @jpirnay are extremely good at jumping on these sorts of things - I would suggest that you should expect a fixed version in a few days at most.