meerk40t/meerk40t #2179: Running on Fedora 38 GLib-GIO-ERROR

jpagliaccio, 2023-10-21:
### Summary Description

I am getting this error when trying to run the pre-compiled Linux version on Fedora 38 - Any suggestions? 

(MeerK40t-Linux:739749): GLib-GIO-ERROR **: 09:22:35.135: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
Trace/breakpoint trap (core dumped)

Thank you. 

### Additional Details

Appears to start/run with  -Z, --gui-suppress (completely suppress gui)

### MeerK40t Version

MeerK40t-Linux:739749

### MeerK40t Type

Executable

### Your Operating System

Linux

tatarize, 2023-10-21:
I'll run it passed the Linux Devs. You could, if you need it to run, just run from source. Install python3 then `pip install meerk40t[all]` and `meerk40t` would work. You may need to use a wheel for the wxPython (though if you want to run with `-Z` you don't actually need wxPython.

tiger12506, 2023-10-22:
Current workaround and discussion when this happened a week or two ago.
The current theory is that Fedora 38 uses Wayland, not X11, but our builds use a wxpython that is compiled against GTK compiled against X11.

[image]

jpagliaccio, 2023-10-22:
Hey thanks so much. I am trying to use the GUI. Now I am getting: 
<pre>
GDK_BACKEND=x11 LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./MeerK40t-Linux 
Failed to extract PIL/_imaging.cpython-36m-x86_64-linux-gnu.so: decompression resulted in return code -1!
</pre> 
The lib /usr/lib64/libstdc++.so.6 exists and is a link to libstdc++.so.6.0.32. 

But the good news is, on Fedora 38 there is a login option for "GNOME Classic on Xorg". Thanks to your Waylend comment above, I just tried that ( with just `./MeerK40t-Linux `) and it works. There are lots of warnings, but the GUI is up and running now.   

Please let me know if you want me to post the warnings.

Ps: Regarding the full Python install on Fedora 38, was finally able to run: 

<pre> pip3 install meerk40t[all] </pre>

... but only after after running `sudo yum -y install wxGTK3-devel.x86_64` which got me past all of the wheel errors. There are some warnings from that too if you want them.   

Thanks again. Going to try some laser cuts next.

closed 2024-10-04 by jpirnay