meerk40t/meerk40t #2203: Is Meerk40t support Marlin firmware?

kpbicka, 2023-10-27:
### Summary Description

I didn't find in case how to connect 3D-printer Laser with Marlin firmware to meerk40t software 

### MeerK40t Version

0.9.2000

### MeerK40t Type

Executable

### Your Operating System

Mac (before M1/Bigsur)

Sophist-UK, 2023-10-27:
Marlin firmware is a one of a number of firmwares that support use of GCODE for lasering.

So really you need to be asking whether MeerK40t supports driving GCODE controllers - which it does in theory, though I am not sure exactly what the status is for that functionality and what the reliability is. 

A complementary use case is would be for MeerK40t to emulate a GCODE controller - AFAIK this functionality has not (yet) been written.

tiger12506, 2023-10-27:
> Marlin firmware is a one of a number of firmwares that support use of GCODE for lasering.
>
> So really you need to be asking whether MeerK40t supports driving GCODE controllers - which it does in [...]

Pretty solid, actually. Tested with FluidNC and GRBL controllers both.

> A complementary use case is would be for MeerK40t to emulate a GCODE controller - AFAIK this functionality has not (yet) been written.

Actually, one does exist. grblcontrol is the gcode controller equivalent of ruidacontrol, because Lightburn also supports controlling grbl lasers. It was my understanding that this works, too.

I'm not certain if there are any tweaks that would need to be made to support Marlin, but I'm sure they would be fairly minimal.

tatarize, 2023-10-27:
Current no.

We support gcode and our gcode support is actually really quite good. You could save out some of our gcode from the normal setup and it would properly work. However most of our interfacing is built around GRBL which is mostly non-compatible with Marlin. The gcode is compatible with either though.

It's a little weird to understand but there's two main parts of any laser cutting control protocols. There's the program code, which is sequential and issued one line at a time, you complete one line, you move on to the next line until you finish the execution. And secondly there's the realtime non-out-of-band non-sequential commands. This usually includes things like `estop`, `pause`, `resume` and in GRBLs case some stuff like speed up, slow down, increase power. But, this is true for *all* laser controllers, and I've reversed engineered a lot of them by this time.

Our gcode support is quite good at this point, mostly because Jpirnay bought me a cheap Ortur2 diode laser, so the weird points got nicely softened. The `grblcontrol` stuff was also mostly finished and I couldn't poke a hole in it.

Now, Marlin will share mostly the same gcode it'll use a few different commands and support some others but we don't stray that far from the standard cut this, cut that, stuff. The real difference in non-support is with the realtime commands, and the interfacing. We added a few buttons, for example, to the controller for GRBL so you can reset the controller and send `clear alarm` commands. You can rightclick the `home` button to send `physical_home` command which is `$H` rather than `G28`. 

So, while we can generate gcode correctly and you could maybe even tell the gcode driver to `ignore greet` requirement and get it to connect to the Marlin over serial port. It wouldn't control the Marlin realtime stuff very well at all. You might be able to get it to work, but its not really supported per se.

There's a few offbrand grbls stuff like FluidNC (which has it's own couple extra realtime commands that also should maybe get support for things like file-send and file-replay stuff. There's also some stuff like repeatedly querying the grbl for status `?` that wouldn't work on Marlin.

I have Marlin on my 3D printer and did test it, some early GRBL stuff with Marlin firmware but there's a lot of realtime control stuff built into meerk40t that wouldn't work at all. You could certainly use the GRBL driver and save out a gcode file which would not include any realitme commands (it's just the series of commands to do the program stuff) and that would likely work in Marlin.

However, since we already have it writing gcode and it just needs to use a well documented protocol it would not be that hard to support Marlin, though it's not well known for it's laser cutting prowess. The open question with supporting Marlin is how much stuff gets reused and at what levels. I assume maybe that we could just reuse the Driver code just does the main program code parts, it would need a different controller and different interface hooks and button within meerk40t. Probably it would need very few since I'm not seeing a lot of realtime stuff inside Marlin.

But, most of the code could probably be reused. The interfacing could also be reused, and I think just remove a bunch of parts of it where it sends other commands or expects grbl-like replies would need to be recoded.

It's an interesting problem since unlike a lot of the drivers, these two are close cousins and write the same general gcode out, but is that a need to duplicate the fullstack and cut it down to size, or are there sections that could be totally reused like the `driver` code there.

tatarize, 2023-10-28:
There's not apparently much if any realtime commands:

https://marlinfw.org/meta/gcode/

Maybe I just haven't found them. They wanted some stuff for abort the job in the github but they don't seem like they implemented any of them.

Sophist-UK, 2023-10-28:
I suspect that the firmware for your specific machine might be a fork of Marlin i.e. based on Marlin but enhanced.

But the bottom line seems to be that there are probably not a lot of Marlin-based laser engravers out there, so there is not going to be a great business case for @tatarize spending the effort to support it (even though if it takes less work, the business case is easier) nor will there be a base of testers to check it works in a methodical manner.

Have you looked at whether there is any alternative firmware available for your controller, or considered swapping to a different controller?

Sophist-UK, 2023-10-28:
Actually I see that there is a [Marlin subforum on the Lightburn forum](https://forum.lightburnsoftware.com/t/about-the-marlin-category/23/3), so it looks like Lightburn supports Marlin - in which case perhaps there are more Marlin-based machines out there than I thought.

tatarize, 2023-10-28:
There are a few, but mostly it's for 3d printers and you just tend to turn them off after the print dies. They don't tend to pause or resume or hard abort as a thing. There's some commands to do things like send progress to the LCD which might be interesting. But, mostly I'd just make a very generic gcode sender. Also, I'm not sure on Marlin's lasermode being that good. I know when I tested gcode stuff on my marlin 3d printer it was pretty clear that the motions stopped a lot.

Marlins also have SD card files. So it could be asked to read the SD card and play files from the SD card. 

I don't think it's that much work since GRBL seems a bit more advanced for lasers and for non-gcode interactions. It seems like Marlins are built more on the idea that when shit goes bad... you have a long time to deal with it, and stop sending code. Which seems a little different than there's a big fire, it's running into the wall, and blinding passersby.

I'd figure an easy first version would be to just copy-pasta the grbl and remove most of the GRBL specific features and stuff. Just send Marlin compatible gcode along the serial connection, or a normal gcode data blob.

tatarize, 2023-11-01:
Also it lacks things for $32=1 sort of laser mode set, but needs some stuff like `G4 I` commands for dynamic power movement stuff so it would seem like it would need a different driver.

kpbicka, 2023-11-01:
> Actually I see that there is a [Marlin subforum on the Lightburn forum](https://forum.lightburnsoftware.com/t/about-the-marlin-category/23/3), so it looks like Lightburn supports Marlin - in which case perhaps there are more Marlin-based machines out there than I thought.

I tried Lightburn software, but it's paid and any business can easily pay $60, but for my personal use at home this price is too high. I found hacked Lighburn, but it's for Windows, and I only have a Mac.

tatarize, 2023-11-01:
Well, I certainly wouldn't advocate software piracy, but the alternative there is actually a bit expensive too (mostly in time and work). I'd need a bit of feedback to make Marlin work as a driver there. I did some of the basic work there (see #2209). But, it's mostly just removing some of the other features of GRBL. Though if you really want I guess I could dig my 3d printer out of the back of the garage, it has an old Marlin controller. Mostly I figure some work to get it to talk and then removing some more features there.

tatarize, 2023-11-01:
I'm fairly sure the gcode blob would want to be made more generic for gcode rather than grbl and the GcodeJob and loader. Most of the rest of the code is pretty exclusive towards gcode. But, it still seems like Marlin in a subpar choice for a laser controller. But, what it lacks in feature it gains in the simplicity of programing features that don't exist. But, the best I can tell it doesn't look like you'd even be able to realtime pause the laser.  It's basically just a glorified gcode writer and sender.

Sophist-UK, 2023-11-01:
> I tried Lightburn software, but it's paid and any business can easily pay $60, but for my personal use at home this price is too high. I found hacked Lighburn, but it's for Windows, and I only have [...]

I wasn't suggesting that you use Lightburn, but rather suggesting that if Lightburn supports Marlin then there must be a reasonable size consumer base of Marlin users.

closed 2024-10-04 by jpirnay