meerk40t/meerk40t #2204: [Bug]: QR Code - errors when attempting to Burn

shawnhenson73, 2023-10-30:
### Summary Description

QR code generates successfully.  
Program crashes when attempting to burn

### Crash logs

```shell
MeerK40t crash log. Version: 0.9.2000 on Windows: Python 3.8.10: AMD64 - wxPython: 4.2.1 msw (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "xml\etree\ElementTree.py", line 1076, in _escape_attrib
TypeError: argument of type 'bool' is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "meerk40t\gui\laserpanel.py", line 495, in on_button_start
    "planz clear copy preprocess validate blob preopt optimize\n"
  File "meerk40t\kernel\context.py", line 32, in __call__
    return self._kernel.console(data)
  File "meerk40t\kernel\kernel.py", line 2194, in console
    # Process command matches.
  File "meerk40t\kernel\kernel.py", line 2241, in _console_parse
    "{command} is not a registered command in this context: {context}"
  File "meerk40t\kernel\functions.py", line 279, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
  File "meerk40t\core\planner.py", line 568, in plan_preprocess
    busy.change(msg=_("Preprocessing"), keep=1)
  File "meerk40t\core\cutplan.py", line 213, in preprocess
    property_op(kernel.root, node)
  File "barcodes\main.py", line 426, in update_qr
  File "barcodes\bcode_logic.py", line 647, in update_qr
  File "barcodes\bcode_logic.py", line 486, in render_qr
  File "qrcode\image\svg.py", line 60, in to_string
  File "xml\etree\ElementTree.py", line 1133, in tostring
  File "xml\etree\ElementTree.py", line 772, in write
  File "xml\etree\ElementTree.py", line 930, in _serialize_xml
  File "xml\etree\ElementTree.py", line 1099, in _escape_attrib
  File "xml\etree\ElementTree.py", line 1053, in _raise_serialization_error
TypeError: cannot serialize True (type bool)

Local variables:
self : <meerk40t.gui.laserpanel.LaserPanel object at 0x12942C40>
event : <wx._core.CommandEvent object at 0x0E6594A8>
busy : <meerk40t.gui.busy.BusyInfo object at 0x0E1D35C8>
plan : z #1 RasterOpNode
```

### MeerK40t Version

Latest as of 10/29

### MeerK40t Type

Executable

### Your Operating System

Windows

shawnhenson73, 2023-10-30:
I did open the same qr code in the previous version and it worked fine.   I thought maybe it was cause I linked it to  Rick Roll video and the software was saying  "Not today"  lol.

jpirnay, 2023-10-30:
Do you have a wordlist pattern as part of the string, so something in curly brackets like ``{date}``?

shawnhenson73, 2023-10-30:
https://www.youtube.com/watch?v=dQw4w9WgXcQ

jpirnay, 2023-10-30:
That is not really helpful... 
Anyway I figured this one out, needs a revision of the barcode plugin (will be published later tonight). If you are using the precompiled packages (Windows exe, Linux / Darwin package) then the next mk release will include the fix, if you are using the source files a ``pip install meerk40t-barcodes==0.2.3`` should be good enough

shawnhenson73, 2023-10-30:
that web link was what I put in to generate the QR code.   it generated .
but when I go to burn is where the software crashed.
Ill switch to the source.
Thank you for looking into it

closed 2023-11-22 by tatarize