meerk40t/meerk40t #2210: [0.8.12] M2 bed size edge case (literally)

Sophist-UK, 2023-11-01:
My bed size is set to 320mmx220mm.

If I try to move to x=320mm or y=220mm I get an error saying outside bed, but I can move to 0.01mm less than this OK.

So this looks like use of < when it should be <=.

tatarize, 2023-11-01:
Should it tho? If your bed is 220mm and you move there, you've hit the edge. I guess it does make infinitesimally more sense to not allow you to move one epsilon further.

Sophist-UK, 2023-11-01:
IMO it should. If I say the bed is 320.00mm wide rather than 319.99mm wide, I would expect it to go right up to 320.00mm.

But now that I think about this, I am reasonably sure that when I set this bed size (in 0.7.x?) I tested it moving to 320.00mm to be sure it wouldn't hit the stops and it worked (and indeed I think when I tried 321.00mm it actually hit the stops). But now it seems to be prevented - was there some extra code added to prevent it from moving outside the bed dimensions in more recent versions?

tatarize, 2023-11-02:
Are we talking bout Move(x,y) in the Move panel? That seems right.

```python
    def source_contains(self, x, y):
        """
        This solves the AABB of the container, not the strict solution. If a view is rotated by a non-tau/4 multiple
        amount, we could generate false positives.
        @param x:
        @param y:
        @return:
        """
        # This solves the AABB of the container, not the strict solution
        x0, y0, x1, y1 = self.source_bbox()
        return x0 <= x <= x1 and y0 <= y <= y1
```

I guess we're specifically talking about that same panel but in 0.8.12?

I totally agree you're right about this. It'd be really hard to set your panel correctly otherwise without all insideness as a thing.

tatarize, 2023-11-02:
While a bit weird, that appears to also be correct. How are you getting this edge condition. It doesn't seem like it's navigation move_to (works in 0.8.12) too

Sophist-UK, 2023-11-09:
Perhaps a floating point rounding error?

tatarize, 2023-11-09:
The spot I looked seemed fine codewise. Can you confirm what dialog doesn't work. The move pane seemed like it was fine. Since it would include the edge. I guess there's the issue where floating points can be non-equal due to slight errors way deep in the fractional part.

Sophist-UK, 2023-11-09:
The use case is simple. Bed size for m2 nano set to 320mm x 220mm. I tried to move to (320,1) and got an error message about being outside the bed.

Since the code uses <= I suspect the only possible cause is floating point rounding. As with ALL floating point rounding, comparisons need to be of the form `min - delta <= x <= max + delta` where delta is a tiny but non-zero positive value of (say - with bed sizes O(2)) 10^-6.

tatarize, 2023-12-07:
What are the direct steps required to recreate this. There seems to be a couple different places and they react slightly differently.

Sophist-UK, 2023-12-07:
Literally set bed size, use the move pane to move to the exact max bed size.

closed 2024-10-04 by jpirnay

Sophist-UK, 2024-10-04:
This probably hasn't been completed.

If you are simply going to mark them as completed without posting the commit or PR which fixes it, this suggests a disregard for minor bugs which cumulatively give a very bad impression about MK. We have all experienced software which works most of the time but where you hit so many minor problems so frequently that you feel the software is very sub-standard.

jpirnay, 2024-10-05:
Probably is not enough, neither are issue reports without evidence. Here are my attempts 310x210mm bed
[image]
Won't be reopened unless evidence could be provided