meerk40t/meerk40t #2225: [0.8.12] Grouping a group node results in left over group

Sophist-UK, 2023-11-05:
When I want to group these nodes (selecting the group rather than the contents):
[image]
I get:
[image]
i.e. there is an empty group node left behind.

I get the same if I have selected from the scene and have the child nodes selected rather than the existing group node:
[image]

I cannot reproduce now, but I also had the existing group end up ungrouped inside the new group. Something different about the selection process I think.

tatarize, 2023-11-09:
What's the correct answer? Clear out all empty groups on the group operation. Or check any parent of the old element and delete them? I'm sure this is a thing, but I don't tend to delete the groups aggressively.

Sophist-UK, 2023-11-09:
In the first example, the entire group is moved inside, so the left-over empty group needs to be deleted.

In the second example, I would argue that for rectangular scene selection, this is actually a selection issue - when you select objects using a rectangle select, then for any groups where all containing elements are selected, the group should be selected also. 

Also in the second example, there are two cases to consider when you group selected items:

1. When all items in the group have been selected (in the tree) but the group itself is not selected - in this case I would personally include the group in the move and leave the contained elements grouped as they were before in  the new group i.e. in the second example above, it would give the same result as the first example above.
2. When only some of the items in a group have been selected then I would personally prefer to see the group recreated - so that there are now two groups each of which contain part of the original contained elements.

(Apologies for the confusing description. Please ask if you need me to clarify.)

closed 2024-10-04 by jpirnay