meerk40t/meerk40t #2234: [0.8.12] Undo after File New goes back one too many steps

Sophist-UK, 2023-11-09:
Original:
[image]

Delete one:
[image]

Delete second:
[image]

Delete third:
[image]

File New:
[image]

Undo:
[image]

File New:
[image]

Undo:
[image]

File New
[image]

Undo:
[image]

tatarize, 2023-11-09:
I think the issue is that file-new is not triggering the a marked undo position. So when you undo, then new, you've consumed that undo, then you failed to force in another undo to fill the old spot.

Sophist-UK, 2023-11-09:
Yes - that was my guess too.

tatarize, 2023-11-17:
The fix to this has a side effect that it creates the undo-point which then can be undone even if nothing in the scene changed. You create a new project 5 times and you have 5 pointless undo points in a row.

Sophist-UK, 2023-11-17:
Of course you could code to check the last command and not add an extra if it is a repeat of the last command and it is on some sort of list of commands where a 2nd entry is the same as the first entry.

Or you could check to see whether the tree is exactly the same as the last version and not add it if the tree hasn't changed.

But not sure it is worth doing either of these.

tatarize, 2023-11-17:
It's weirdly just on the wrong side. It's harder to see if the tree is changed. But, is doable. Just measure how many ops and elems were there before and mark it if equal to zero. The other oddity is the fix was on the wrong side of the work. It flagged before the clear which seems right but isn't. The flag is for the current undo state and preserves it.

We added undo/redo highlighting so it's a bit easier to see this stuff.

tatarize, 2023-11-17:
We're just about ready for 0.9.3 so I'm down to making sure anything that isn't quite fixed is dealt with.

Sophist-UK, 2023-11-17:
Fix the UX issues before 0.9.3 because that is pretty much all that is wrong with it - and a lot more people will switch once they are fixed.

tatarize, 2023-11-17:
I don't usually fiddle that much with UX. The biggest UX change was the ribbonbar change which is not something I would delve into again without more specifics and I haven't been monitoring that that much.

0.9.2 has a problem where it crashes a bunch. So I'm looking forward to a new version. So UX mostly unchanged it needs another full version in short order. Its simply not a large enough hurdle to prevent the new version which already got delayed twice due to GRBL issues.

tatarize, 2023-11-17:
I'll close it myself when 0.9.3 gets published. (though I do like those auto-close features)

tatarize, 2023-11-17:
Though I am otherwise done and waiting, I suppose, for other people to sign off on main being 0.9.3 so I'll read through the thread.

Sophist-UK, 2023-11-17:
> I don't usually fiddle that much with UX. The biggest UX change was the ribbonbar change which is not something I would delve into again without more specifics and I haven't been monitoring that that much.

I do understand that you like working on functional code and not UX. But that doesn't mean that UX code can be ignored. 

As I have said countless times before, you need more help than just @jpirnay, not only in code development (esp. for areas that you are less interested in) but also for documentation and community support. If you want to attract **and keep** people to help you, then you need to make the community more pleasant (and if you don't want to have to spend time on developing and enforcing a code of conduct then find someone who has the ethics and give them the responsibility for making MK development community a welcoming place).

> 0.9.2 has a problem where it crashes a bunch. So I'm looking forward to a new version. So UX mostly unchanged it needs another full version in short order. Its simply not a large enough hurdle to prevent the new version which already got delayed twice due to GRBL issues.

Full releases need to be stable and deliver something significant. If 0.9.2 isn't stable then it should **NOT** have been released as a full release, but as a beta instead. And now you want to rush out a 0.9.3 release in order to address the instability of 0.9.2 despite there being outstanding UX issues which prevent it being a fully rounded release?

A few more days to allow Jens to finish the UX issues and get a fully rounded 0.9.3 should not be a concern.

Sophist-UK, 2023-11-17:
> I don't fully believe that some bits slightly overlapping hinders the functionality of that work

If I read this correctly, you are effectively saying that you don't consider UX issues to be **un**important. Whilst this may be true for you personally as an individual, this is simply **NOT** the case for the Community as a whole. UX issues have a significant impact on how MK is perceived by others, and if you want the widest possible use for MK so that your efforts benefit the most people, then you (or the team) need to give UX issues some reasonable priority.

Which is **exactly** why I made the previous statement (again) about the need for a team - I know that UX is not your thing, and that is why you need more people involved.

Clearly you are never going to be convinced of this by me restating it - perhaps @jpirnay can convince you.

tatarize, 2023-11-17:
It's importance, I just don't think it's that bad and I actually sort of like it. Most of the UX things you raised have been checked off by Jpirnay. The current UX stuff seems pretty solid. And it's a bit more than just GUI. And if you have specific requests I'd be happy to address them. 

UX-wise the biggest thing in the pipeline is making the laser-panel a lot more configurable so a lot of the device specific buttons and functions that currently end up in the ribbonbar will end up in the laser-panel.

And some of the changes needed are more sweeping changes towards a more wholistic interface. Since a lot of times the development of a laser project involves designing, burning, and adjusting, in a much more connected approach. So we tend to laser things then adjust the stuff we're lasering, canceling out remaining passes or speeding them up, etc to get the settings right, while going back and adjusting other things. It's less printing things on wood, and more of collaboration in building between you and the laser. (Usually, sometimes it's cutting 1000x of these dohickies).

I think font overlapping is pretty minor in the grander scheme and it needs to be a lot better integrating into various workflows. Sloppier buttons doesn't really play that big of role in UX. It needs to be sleeker, and more focused, and built around a much more holistic view of laser projects.  Though making every button that currently does nothing disabled, is a bit of that, or making sure the code for various sizes of icons makes sense is important. I think there's some issues that are more important. 

I thought most of the UX list of things was a bit meh, when really what's needed is somebody with real design chops answering the question: what set of tools would provide you the best laser experience? And some of my philosophy here has moved towards the idea of tools, like if you're doing some work, what sorts of functions would help you with that work.

... that and I was thinking about buying a joystick and making it zip the grbl device around. Also, the current last release crashes a whole bunch and needs a version that doesn't.

Sophist-UK, 2023-11-17:
> I think font overlapping is pretty minor in the grander scheme

> I thought most of the UX list of things was a bit meh

Yes a lot of what I reported was really very minor - but if no one reports the minor things they **literally** never get addressed, and the trouble with UX is that a lot of the time a bunch of things that are minor on their own when taken together create a much worse impression.

As a genuine real-life example from this morning, my wife used 0.9.3b2 for the first time (it's actually her laser for her craft business - I am just her woefully underpaid laser technician!!) and her very first comment was that the icons on the big Go/Pause/Cancel buttons were broken and was this new version reliable. In other words, tiny UI things can create a vastly disproportionately bad impression.

tatarize, 2023-11-17:
Yeah, minor issues do need reporting. And yes, many minor issues makes for an overall feel that is negative when a lot of small things are wrong. We all agree they are important, and small things are immediate. It's one thing to say I'm dealing with the overall philosophy of laser software, it's another thing when the icons are weirdly cropped and look bad. 'You do the small things like keep the boat floating I do the big things, sitting on the boat and pondering the meaning of the universe.'

My comments weren't that UX is unimportant but they do, if they haven't gotten notably worse, take a back seat to the UX of having the program crash a bunch like the last full release was actively doing. As far as UX, I did pitch writing our own buttons to get some OSX stuff to be consistent (black background buttons on OSX sit on a white square so coloring the button icon white makes them invisible).

Another aspect may be sour grapes. I'm not that good at it, so it must not be that important.

Sophist-UK, 2023-11-18:
My comments were not intended as a criticism. We all have things we like doing (like "pondering the meaning of the universe and how to create a brilliant kernel") and stuff we don't like doing (e.g. UX tweaks) - and goodness knows I have difficulty dealing with my own end-of-project small stuff and snagging lists.

But this is why you need to foster a team spirit and grow a team. So you can delegate the minor stuff and the stuff you would rather not have to do.

As for crashing, yes of course crashes need to be fixed. But more importantly you need to stop pushing out full releases that crash!!! Which means **NOT** rushing into creating full releases until they have had time to be tested. If people are experiencing crashes, give them a bug fix beta!!

tatarize, 2023-11-18:
I have no qualms with criticism. They taken as a whole a series of issues really does give the user a bad experience. And it matters.

The crash was an 11th hour change after most of the testing and snuck in, really our error was not issuing a `post1` released right away. An event we're trying to avoid again in the future by having a branch for any bug fixes waiting in the wings. If any bug fix is made we'll cherry-pick it into that branch and if they are serious enough we'll push a new version. 

As far as UX goes, I think the biggest issues would be had during the initial startup. If you launch the software with a `-X` nukes the settings by refusing to load them at start (but still saving them at close). Is that UX actually the most right for the most people? Just first impressions UX is important. I have a longer standing project to make the user choose their laser from the laser list if they have no laser setup. But, there's also a lot of cool features, effects, and options and maybe their initial states are wrong. Generally if we add a new feature and an option for that feature, `off` is a lot safer and better than on.

Sophist-UK, 2023-11-18:
I would suggest that you start a thread on the FB group asking people to post screenshots of their MK pane layout and whether they consider themselves to be a novice or experienced user of MK.

When you have a big selection of these to review, you will be able to analyse how many are still using the default view, and for those who have modified their screen layout which panes get used most and (not so much where they are on screen but) how they are grouped together.

You can then try a new default layout based on this analysis.

That said, I still think that thinking out the user lifecycle and the stages they go through (design, burn planning, burning) and having the UI provide tabs for the user to select which then change the panes layout might be a good way forward.

Implementation ideas:

1. Allow user to define several pane layouts in the same way that you can currently have several ribbon layouts, (Not the same Preferences UI, but the same concept.)

2. Decouple the Ribbon tabs from the Ribbon - and make each tab change both the Ribbon layout (as currently) and the pane layout (new functionality).

3. Provide an overarching UI set capability, allowing you to deliver several (and fixed) alternative pre-created layouts and for the user to select these, and to save their own modified layouts to the list. You can then populate this with a) the current layout, B) provide a legacy layout which looks identical to the current default one, and C) a new default layout which might be based on a lifecycle model and also use the analysis above.

tatarize, 2023-11-18:
Okay. Admittedly we did it again... but this time fixed it instantly.

https://github.com/meerk40t/meerk40t/commit/dae117b6520165810570a7117c39385c606a59a7

The 0.9.3000 decheck the `Optimize` checkbox crashes the program. So I fixed it, nuked the version, and flagged the message on the parse to notify the user that it's fixed in 0.9.3-post1

Sophist-UK, 2023-11-18:
You will keep doing this until you do the following:

1. Publish a pre-release beta which will become the full release UNCHANGED. If you find bugs, then you fix them and publish a new pre-release beta and you iterate until there are no more bugs reported and then you publish a full release which is **identical** to the last pre-release beta.

2. You deliver new functionality in early-betas and get feedback - and eventually you decide that there is enough new functionality for a new full release and you go back to step 1.

You should IMO **NEVER** issue a full release that has not had A) decent testing from the dev team, and B) some real-life testing from expert users.

I didn't get time today to publish a UX review of 9.3 but suffice it to say that the Ribbon/Toolbar changes were largely good with a few minor niggles, but some other scaling changes were a nasty disaster (and looked like they hadn't had ANY testing), and some UX issues remain unfixed and ugly!!!! 

SUMMARY: The 9.3 UX is **still** not ready for production IMO - and today my production use went back from 0.9.2b2 to 0.8.12 simply because the UX is broken significantly enough for me to consider it impractical to use.

commit d1658607 (2023-11-17): Fix #2234 with corrected undo mark
commit 76d6b453 (2023-11-17): Merge pull request #2266 from meerk40t/undo-indicator

Internalize Undo Signals, Fix #2234
closed 2023-11-22 by tatarize