meerk40t/meerk40t #2237: [Suggestion]: More versatile coordinate system

Whateverman2012, 2023-11-10:
### Summary Description

Hi,

  I have a sculpfun laser diode and wonder if it is possible to have an option to define origin, x and y axis directions of the canvas.
  In my particular case, I need the origin to be at bottom-left of the canvas.

Regards

### Current functionality

Actually the canvas is oriented for Co2 lasers.
The origin is often far and at left of the user when facing the cabinet.
Actually there are only 2 corners available to set the coordinate system of the canvas.

### Future functionality

With laser diode (grbl) cutters homing is more frequently near the user.
At left or at right facing the cabinet.

There is already an option to swap xy but a more generic option would be to have both x and y flip as independent choices.
Which would allow having the origin in one of the 4 corners of the canvas.

### MeerK40t Version

0.9

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-11-11:
This sounds like a typical default GRBL setup.

Swap XY doesn't do Flip-x and Flip-y. There are independent check boxes for that right above that axis.

[image]

The `Flip Axis` is right above SwapXY. SwapXY actually swaps X and Y. So jogging right will jog down. This matters because sometimes the steppers for X and Y are switched, this setting isn't typically used.  GRBLs are usually just Y-flipped. 

This typically also changes the origin since -X heading towards 0 and -Y heading towards 0 is dictated by the stepper directions.

There's certainly options for a bunch more robustness in this, galvos, for example, could easily need to be rotated towards what the user considers the `top`. And there's some questions about user origins that are independent of device origins. But, this appears to be a somewhat standard device that needs `Flip Y` checked and no other coordinate changes.

Whateverman2012, 2023-11-11:
When I use flip or swap nothing is changing in the gui regarding to the rulers nor the origin (red-green axis).

Does it mean that those actual options does not change the visual aspect (gui canvas) but are more related to the hardware layer ?

Regards

tatarize, 2023-11-17:
The gui rules wouldn't change. The red-green might not either. The red dot should change.

Those aspects refer to user-space rather than the mapping between scene-space and device-space. You can also look at the position of the reticles. I may have to finish up a lot of the work here, but I'm quite sure the laser coordinate system is actually moving.

closed 2024-10-04 by jpirnay