meerk40t/meerk40t #2242: [Suggestion] Zoom to Selected / Move to Corner of selected when nothing selected

Sophist-UK, 2023-11-12:
Some user actions need objects selected, and the examples I can think of are `Zoom to Selected` and `Move head to corner / centre of selected`.

With nothing selected:

* Currently: Nothing happens
* Proposal: MK assumes that everything is selected - or perhaps assumes that everything in Operations is selected, the difference being those occasions where you have deliberately omitted some elements from Operations.

tatarize, 2023-11-12:
I'm a fan of being overt here. If they didn't select anything. Nothing is the correct answer. Though Zoom should maybe zoom to bed if nothing is selected. But, move laser head should absolute not do a thing based on a fallback. It could hit walls or damage something maybe if it does unexpected things.

Sophist-UK, 2023-11-12:
The correct answer is to do what the user expects and not what you think he expects.

IMO if the user does a Zoom to Selected, they expect a Zoom to happen. If they do a move to corner/centre, they expect a move to happen.

If you analyse the three options:

* If you **don't do** the action, you force them to select the objects and command the action again, and the user is happy on the second attempt.
* If you **guess** the action but **get it wrong**, you force them to select the objects and command the action again, and the user is happy on the second attempt.
* If you **guess** the action and **get it right**, then the user is happy on the first attempt.

In other words, if you guess in these situations there is definitely no down-side to guessing, and in many cases a definite up-side.

Moreover, since if the user is expecting it to zoom or move to subset, more than likely they will have explicitly selected that subset before issuing the zoom/move - so if MK uses the whole design as a default, IMO it will be right far more often than wrong.

In other words, where there is no downside, there is nothing wrong with the software making an educated guess.

tatarize, 2023-11-17:
Jpirnay's solution was to turn off the button if there was nothing for the button to do.

Sophist-UK, 2023-11-17:
> Jpirnay's solution was to turn off the button if there was nothing for the button to do.
> 
> (It hadn't really crossed any of the rest of our minds, but it's far and away the best solution.)

This is a better solution than doing nothing, but IMO not as good as guessing (providing that there is no down side from guessing i.e. no more effort to correct a guess than correcting a do-nothing response.

closed 2024-10-04 by jpirnay