meerk40t/meerk40t #2243: [0.9.3b1] UX review

Sophist-UK, 2023-11-12:
So far I have been continuing to use 0.8.x for production due to UX issues, but I really, really want to switch over to 0.9.x soon.

I took a look at the latest beta today, and thought that a UX review might help to iron out all the UX things that are still an issue for me.

Firstly I must give a huge plaudit for the high-res, vector graphics. The UI looks EXTREMELY crisp, and the Tools pane now works great in a horizontal panel. 

Now to the UX issues, all of which I expect will mostly be easy to fix:

#### Pane content scaling
Some panes do not have content scale-to-fit at all (e.g. Drag, Jog), and some have scale-to-fit which don't work well (e.g. Go, Pause, Cancel). Scale-to-fit needs to respect the pane boundaries in both dimensions and be consistent across all panes (or at least those where scale-to-fit makes sense). (It may be fixed when you address Panes, but the same issues also apply to some Tools windows.)

#### Laser-control pane
[image]
Aside from the obvious scaling of the limit-to-bed-size icon...

* I prefer the 0.8 icons for limit-to-bed-size - I don't understand what these icons represent and they do not make clear what option is selected.
* Resizing the pane results in image corruption:
   [image]
* Jog-tab: No scale to fit and (when floating) scroll bars.
* Laser tab: If only one device configured, the device dropdown should be limited as it serves no purpose
* Move tab: When reducing width of pane, when numeric boxes reach a reasonable minimum size, the saved-position grid should be scaled instead of continuing to reduce the numeric boxes area to width 0px.

#### Position pane

* The new 3x3 grid needs tooltips
* The 3x3 grid is way way too big - can we make the whole the same size as the apply-changes button and place it to the right?
* "Keep ratio" should IMO be "Keep aspect ratio"
* Numeric input boxes: Can have a lower minimum size as even with d3.2 there is a lot of space left to the right.

#### Icon and text sizing

Fineness: I suspect that for users who have sight difficulties, the very thin lines may have become difficult to read. The UI might benefit from a vector line-width preference option to allow vector icon lines to be thickened. Similarly some text is very small (Ribbon tab labels for example). I am not sure whether it would make sense to adjust sizes using the Windows 10 display settings text-scaling value or not.

Preferences - GUI tab: I have no idea what the Icon size dropdown now does - I suspect that it may impact the non-vector icons, but I tried to e.g. set the size to tiny and by comparison to the vector icons they were still huge.

Vector icons: The icons and their text labels do **not** grow to fill the available space, only shrink - they are sort of OK for me at their maximum size on my Surface Pro 3 (2160x1440, text at 175%) but some might find them too small. I am not sure whether it would make sense to also adjust these sizes using the Windows 10 display settings text-scaling value.

# Old vs. New
#### Old
[image]
#### New
[image]

tatarize, 2023-11-13:
* 0.9.2002 is the latest version circa yesterday or so (some slight tweaks may be included)
* The expectation is that the tools menu would be placed on the side and tend to run down from the left side.
* Reset might better organize that. But, it would be perfectly acceptable to run them like that, after all user's choice. 
* The icons for the big buttons is apparently based on width rather than the minimum of width or height.
* The position thing seems a bit more crushed. I'm not entire sure what changed there other than icon sizes of the other stuff.
* Everything was switched to vector icons so some icons may have have changed a bit.

tatarize, 2023-11-13:
Unmarked from #2154 

- [ ] No colour background
- [ ] Routine to assign space to panels does not make use of additional space vertically / horizontally to reduce the need for the 'additional buttons' indicator

tatarize, 2023-11-13:
Also, I'm not entirely sure what could cause weaker UX other than the switch over to the ribbonbar which I mean I could pour more time into but I'd probably rather not. Though I am actually sort of planning on replacing the buttons we have on things like the laser-panel with our own type of similar buttons after we had some issues with Mac having their own buttons and having some limited abilities to get the colors right. Though from the ribbonbar I know that's a can of worms and have certainly held off on it.

Sophist-UK, 2023-11-13:
> * 0.9.2002 is the latest version circa yesterday or so (some slight tweaks may be included)

Not available when I started my tests otherwise I would have used it.

> * The expectation is that the tools menu would be placed on the side and tend to run down from the left side.

This is the difficulty with UX - different users have different expectations, and it can be VERY difficult for a developer to know just what users might want to do - so you have to try to anticipate which in the end comes down to either A) you prevent it being put into a horizontal panel, or B) you fully support it in a horizontal panel, which means testing it - no half way measures.

> * Reset might better organize that. But, it would be perfectly acceptable to run them like that, after all user's choice.

No idea what you are referring to here.

> * The icons for the big buttons is apparently based on width rather than the minimum of width or height.

That's what I guessed.

> * The position thing seems a bit more crushed. I'm not entire sure what changed there other than icon sizes of the other stuff.

No idea what this means.

> * Everything was switched to vector icons so some icons may have have changed a bit.

The keep inside bed icon has changed completely and neither icon suggests to me what the button does or what its state is (unlike the 0.8 icons on this button which do make it clear).

Sophist-UK, 2023-11-13:
> * [ ]  No colour background

Having used 0.9.3b1, I now rate this as less important than before (more nice to have than essential), however I think it should be considered as a UX/UI issue on how to use colour to assist the user by differentiating functional areas. So perhaps different colours as background to each different ribbon bar tab. Care also needed to stop it becoming garish.

Sophist-UK, 2023-11-13:
> ... Mac having their own buttons and having some limited abilities to get the colors right.

There are two schools of thought for Windows / Mac x-platform apps. Either you have a philosophy to support the UI Design standards of each platform (**which is a nightmare**) or your philosophy is to have a consistent UI for the app regardless of the platform UI standards (**which is a nightmare**). (Or just not bother with multiple platforms - just kidding!!)

jpirnay, 2023-11-14:
- [x] **B** Huge fence icon 
- [ ] **R** Previous icon constrain movement - _The previous icon was a fully bitmapped one in a poor quality, so I needed to choose a different one (which I personally like better, but that's a subjective thing probably)_
- [ ] **R** Scalable icons in jog/move etc.
- [ ] **?** Image corruption - _Sounds like a graphic card driver issue, I have never encountered it_
- [ ] **R** Laserpane: if only one device active reduce / suppress
- [ ] **R** Minimum size for numerical entry text boxes
- [x] **B** 3x3 grid needs tooltip
- [x] **B** 3x3 grid way too big
- [ ] **R** Keep ratio -> keep aspect ratio - _Already crammed, tooltip uses 'aspect ratio' term_
- [ ] **R** Icon stroke width
- [ ] **B** Inconsistent use/recognition of GUI preference setting
- [ ] **B** Text in icons in toolbar too big 
- [x] **B** Icon in Go/Pause/Stop/Home panel cropped - _Maybe something in the version that was tested, latest version acts as supposed_
- [x]**R** No colour background - _Selectable if you right click on an empty space in the toolbar_
- [ ] **R** Routine to assign space to panels does not make use of additional space vertically / horizontally to reduce the need for the 'additional buttons' indicator

B: Bug, R: Improvement / Feature Request, ?: Unclear

Sophist-UK, 2023-11-14:
> * [ ]  **R** Previous icon constrain movement - _The previous icon was a fully bitmapped one in a poor quality, so I needed to choose a different one (which I personally like better, but that's a subjective thing probably)_

Vector icons cannot be that difficult to create - so creating a vector version of an existing icon can't be that hard. That said, I am not bothered what the icon is so long as it actually indicates what it means rather than being meaningless.

> * [ ]  **R** Scalable icons in jog/move etc.

Whether this is a bug or a feature request depends on whether you think a lack of consistency in the UI is a bug or not. (Personally I do. I do not think that consistency should be optional and a feature **request** is something that can be turned down i.e. it is optional.)

> * [ ]  **?** Image corruption - _Sounds like a graphic card driver issue, I have never encountered it.

It could be a feature (or rather a bug) in my NVIDIA or Intel graphics driver, but I only ever saw it in this specific window/pane - IMO it was more indicative of the icons having a transparent background and the background not being cleared when the icons are moved.

> * [x]  **B** 3x3 grid way too big

I am not sure that this is a bug  as it is ugly and awkward but not inconsistent. But you've fixed it so the point is moot.

> * [ ]  **R** Keep ratio -> keep aspect ratio - _Already crammed, tooltip uses 'aspect ratio' term_

This is minor - I can live without it.

> * [ ]  **R** Icon stroke width

Accessibility is a whole area of expertise - but I suspect that fully blind users won't be doing lasering. But someone who has difficulty focusing might.

> * [ ]  **B** Inconsistent use/recognition of GUI preference setting

See - inconsistency **IS** a bug.

However, if all icons are vector and scale to fit the size of the area they have, we probably don't need this GUI preference for Icon Size at all.

> * [ ]  **B** Text in icons in toolbar too big

Did I really say this? The text and the icons fail to get bigger when they should. There is definitely NOT a problem with text or icons being too bug.

> * [x]  **R** No colour background - _Selectable if you right click on an empty space in the toolbar_

I'll try this in the next release.

> * [ ]  **R** Routine to assign space to panels does not make use of additional space vertically / horizontally to reduce the need for the 'additional buttons' indicator.

Not sure this is quite what I meant. And not sure whether this is exactly what is happening under the covers.

> * [x]  **R** The icons and their text labels do not grow to fill the available space - _Will grow up to 150 pixel now_

I'll try this in the next release.

> * [x]  **R** Ribbon tab labels are very small on High-DPI monitor under Windows

I'll try this in the next release.

> * [ ]  **R** 'Restart needed' fatures should offer an immediate restart

It could be argued that this is an inconsistency with the O/S and so a bug not a feature request. [Just kidding!!]

jpirnay, 2023-11-14:
> Vector icons cannot be that difficult to create - so creating a vector version of an existing icon can't be that hard. That said, I am not bothered what the icon is so long as it actually indicates what it means rather than being meaningless.

The laser is enclosed, fenced in... so it has a meaning ;-) Seriously: if anyone can come up with better icons - happy to look at them and include them.

Sophist-UK, 2023-11-14:
Hmmm - now that you have explained it, I can see the relevance of that icon, but not the other one. Please make yourself available 24/7/365 for eternity to explain this to other users. :-)

More seriously, icons need to be immediately obvious to the majority of people and whilst this was "obvious" to you, it wasn't obvious to me (and I am an experienced user) and likely won't be obvious to many others.

I can easily come up with better icons - the ones in use in 0.8.x!!!! Just not vector versions as yet.

jpirnay, 2023-11-15:
> It could be a feature (or rather a bug) in my NVIDIA or Intel graphics driver, but I only ever saw it in this specific window/pane - IMO it was more indicative of the icons having a transparent background and the background not being cleared when the icons are moved.

I didn't face it for ages, but then I  witnessed it myself just twice - so I confirm the bug status. I assume if icons are been requested too fast to come up with a new size not seen before (eg during resizing), that the generation routine is called  multiple times in parallel. Will try a ``Threading.lock``

Sophist-UK, 2023-11-16:
Another example from the Simulation panel this time.

[image]

tatarize, 2023-11-16:
I'm pretty impressed how well the simulation panel scales there.
Without enough room it does start looking pretty ragged.

Sophist-UK, 2023-11-16:
Yes indeed - it is only the button in the BR corner that doesn't scale well'

Sophist-UK, 2023-11-17:
New issue - Ribbon and Toolbar shrink too late to keep text visible. As you reduce the height of the Ribbon or Toolbar, the text disappears and the icons/text only get smaller once the icon cannot fit. Since you can show or not show the text in MK preferences, both the icon **and** the text should be used to determine when the icons/text need to shrink or can grow.

As an aside, can we have a show/no-show of labels as a right-click context menu for the Ribbon / Toolbar?

tatarize, 2023-11-17:
YEah, I've noticed that too. It seems pretty nice but it is weirdly keeping the ribbonbar and icons just enough so they slightly overlap.

tatarize, 2023-11-17:
Yeah, probably. I know it can be updated pretty easily in the settings for the toolbar there. I'm not sure that it would be needed enough to warrant the speed that a menu-item would provide or worth the potential clutter. These little tradeoffs are why I suck at UI.

[image]

closed 2024-10-04 by jpirnay