meerk40t/meerk40t #2246: [Bug]: Text stroke doesn't seem to be respected?

jaredly, 2023-11-14:
### Summary Description

Here's some text with a copy of the text, having `stroke-width=4` and `stroke=white`, allowing it to stand out from a background
[image]

Unfortunately, on import, the text stroke fails to render, and the text is unable to stand out from the background 🙁
[image]

The Properties dialog for the background text object does show that the stroke is set to `#ffffff`, but I don't see a place to indicate the stroke width?

### MeerK40t Version

875e3037efcfa3b5b0a584ce616a40f1012b4a1c

### MeerK40t Type

Git

### Your Operating System

Mac (M1/Bigsur)

tatarize, 2023-11-14:
Well stroke-width can exist. 
[image]

I don't think properly that stroke exists for pure old raster-text objects. They are technically only filled. 

Vector text has it exist in properties.

[image]

jaredly, 2023-11-14:
Oh interesting, maybe the import needs to be changed then? It's from an SVG `<text stroke="white" stroke-width="4" ...`
Also, how did you manage to set a stroke on that vector text object? I don't see a place for it:
[image]

jaredly, 2023-11-14:
ohhhh ok I understand. Normal "TextNode" elements don't currently support rendering with stroke, you need to create special "vector text" nodes, where you specify the font manually. That's a shame 😬

Sophist-UK, 2023-11-14:
1. SVG text objects are actually very difficult - they can have spans and be curved onto a path. So MK's text support is limited and the general recommendation is to "convert text to path" in e.g. Inkscape.

2. However, I am not sure that this would help in this instance, because a thick white stroke on a path is not going to be rastered. In MK strokes are not rastered, but as this example shows maybe they should be. Inkscape has a function to convert a wide stroke to a filled path, so having converted text to path you would then need to use this function to make the text strokes into a filled path that MK will burn correctly.

3. In any case, MK treats each operation separately, and it treats every image separately even if in the same operation (which maybe it also shouldn't in order to be consistent with raster elements). Where White raster elements overlay Black raster elements in the same operation, then they mask out the underlying blackness and leave an unburned area, but because separate operations are independent, they don't. So the text (converted to a path) will be a raster element, and the map is (presumably) an image, and so these will be in separate operations - which means the white text background will not stop the world image below from being burned. So **if** the map is a vector, then ensure that the map and text are in the same operation; if the map is an image then the only solution is to use an image editor put the text over the map and save it as an image.

jaredly, 2023-11-14:
Very good to know. Fortunately, the entire map is vector, so I can have them be the same operation. But to clarify -- when you say "in MK strokes are not rastered", I don't think that matches my experience. For example, the black vertical line in the bottom left of the image is a path with a stroke, and I believe I've seen such paths included as part of a raster operation. I could be mistaken though? I'm just returning to this map-engraving project after a several month haitus, and it's possible I ended up using k40w for the whole thing last time.

jpirnay, 2023-11-14:
I assume you are using a rather recent version of mk? In preferences you have an option to ask what mk should do if it encountered unsupported /not fully supported svg elements: it can send the file to inkscape asking it to convert these (which is exactly what k40w does, but without letting the user know)