meerk40t/meerk40t #2261: [Suggestion] No redo function

Sophist-UK, 2023-11-16:
If I am `undo`ing several changes (which works GREAT) and I undo one or two steps too far, I would like to `redo`.

In the array you use to hold the history, can you do an undo by moving a pointer back from the end, allowing redo to move the pointer forward again, and when you change something then truncate the array and append the latest change?

tatarize, 2023-11-17:
Why doesn't that already work? Does the `redo` button not work?

tatarize, 2023-11-17:
[image]

vs. 

[image]

Sophist-UK, 2023-11-17:
Oops. Sorry.

closed 2023-11-17 by Sophist-UK