meerk40t/meerk40t #2272: Suggestion: File node context menu "Open containing folder in file explorer"

Sophist-UK, 2023-11-19:
Right click on a file node, select this new menu item and it will open the folder in the o/s file explorer app with the specific file highlighted.

tatarize, 2023-11-19:
Do other OSes allow this? I know we can easily do this in windows by merely running `explorer <dir>` or even `dir` but I don't know if other OSes have this sort of thing. And I might need to check if the whole `highlight this file in explorer` works since that's also focusing on the file in question is pretty helpful there too.

Chatgpt suggested.
```python
def open_file_in_explorer(file_path):
    system_platform = platform.system()

    if system_platform == "Windows":
        # Use the "start" command to open the file explorer in Windows
        subprocess.run(["start", "explorer", "/select,", os.path.normpath(file_path)], shell=True)

    elif system_platform == "Darwin":
        # Use the "open" command to open Finder on macOS
        subprocess.run(["open", "-R", os.path.normpath(file_path)])

    elif system_platform == "Linux":
        # Use the "xdg-open" command to open the file explorer on Linux
        subprocess.run(["xdg-open", os.path.normpath(file_path)])

    else:
        print("Unsupported operating system")
```

tatarize, 2023-11-19:
Tested it in windows and it worked. Not sure about OSX or Linux.

Sophist-UK, 2023-11-19:
You can definitely select the file in Windows - I think you do "explorer.exe /select C:\Meerk40t\myfile.svg".

But you might want to look at this package https://pypi.org/project/show-in-file-manager/ which claims to do this cross-platform.

tatarize, 2023-11-22:
Did it cross platform. Mac worked fine. Linux needed to launch the directory and thus not focus the file.

closed 2023-12-17 by tatarize