meerk40t/meerk40t #2275: [Bug]: Fillets are lost in DXG files

jpagliaccio, 2023-11-20:
### Summary Description

Fillets are lost when opening DXF files created by LibreCAD. Tried DXF formats 2007 thru R12 with the same result. 

### Additional Details

The same file will open properly in Inkscape but there is an import note about extrusions when opening DXF with fillets. LibreCAD 2.2.0.2 and Meerk40t v0.9.3000.  Bevels open properly.  

### MeerK40t Version

v0.9.3000

### MeerK40t Type

Executable

### Your Operating System

Linux

jpirnay, 2023-11-20:
Not sure I follow - trying it with a simple rectangle myself, here's what mk imported:
[image]
Can you share the culprit file, please?

Sophist-UK, 2023-11-20:
No attachment because you sent it as an email reply rather than using a browser to attach it. And since a picture is worth a thousand words, some pictures of the DXF in the originating program and when it is loaded into Meerk40t would help.

tatarize, 2023-11-21:
Yeah, I might need the minimum example of this, since DXF allows so many ways to make arcs.

jpagliaccio, 2023-11-21:
Tatarize,

In LibreCAD 2.2.02
1) File, New
2) Line, 2 Points
3) Draw some lines making corners
(tip: right click for a recently used menu)

[image: image.png]
4) Tools, Modify, Fillet
5) Enter Radius, (eg: 20)
[image: image.png]
6) Select the first line (towards the corner)
7) Select the second line (towards the corner)

[image: image.png]
8) Save
Open the file in meerk40t

PS: I tried the Freehand tool to draw arcs and those don't work either.

Regards,
Joe

Sophist-UK, 2023-11-21:
@jpagliaccio I repeat - PLEASE use a web browser on this page to comment on this issue so that your attachments get added!!

jpirnay, 2023-11-21:
Created in LibreCad 2.2
[image]

Meerk40t after file load
[image]

jpagliaccio, 2023-11-21:
LibreCAD 
1) 2 Lines
[image]
2) Fillet 
[image]
3) Meerkt40 
[image]

jpirnay, 2023-11-21:
Can you please confirm your version? This bug was fixed with the full 0.9.3 version - https://github.com/meerk40t/meerk40t/releases/tag/0.9.3001

jpagliaccio, 2023-11-22:
MeerK40t v0.9.3001
Python 3.6.8
wxPython 4.1.1 gtk3 (phoenix) wxWidgets 3.1.5
The pre-compiled MeerK40t-Linux version from https://github.com/meerk40t/meerk40t/releases
I just did a fresh download and tested. Same result (no fillets).

jpirnay, 2023-11-22:
Again, my tests are not able to reproduce this, so I would need an example file that demonstrates the issue, please.

Sophist-UK, 2023-11-22:
Jens, if @jpagliaccio doesn't upload his example file so that you can reproduce this issue, I think you might have to closed it as unreproducible.

tatarize, 2023-11-22:
We need an example. You could email me at tatarize at gmail if you want, or actually drop it to the issue. We don't have a copy of this, but it's entirely possible that it's using an unsupported dxf thing to do the fillets and we just need to extend the support but we'd need an example to check that.

jpagliaccio, 2023-11-23:
Emailed the file.

tatarize, 2023-11-24:
Nothing received...Checked spam etc.

jpirnay, 2023-11-25:
Hmm, I need to flip it, but then I have:
[image]

jpirnay, 2023-11-25:
So according to the latest code:
- [x] fillets work as they should
- [ ] text is placed correctly

So, a different bug...

jpirnay, 2023-11-25:
The text issue (plus some issues around correct image import) will be dealt with under  PR#2302

closed 2024-01-15 by tatarize