meerk40t/meerk40t #2293: [Bug]: Window formatting contents incorrectly

jpagliaccio, 2023-11-23:
### Summary Description

Window formatting contents (gadgets) incorrectly. For example the Navigation pop-out window contents are cropped no matter the window size. Many Gtk warnings are produced for width and size. The Navigation window produces the most Gtk warnings.   

### Additional Details

For example: 

[image]

In my shell window there are many repeated Gtk width warnings (eg: Negative content width -9).

### Crash logs

```shell
-------------------------------------------------------
Clicking the Navigate Button (a subset of the output)
-------------------------------------------------------
(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.475: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.510: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.510: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.539: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.539: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.573: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.573: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.624: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.624: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.656: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 08:56:28.657: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)
---------------------------------
Clicking Simulate Window Button 
--------------------------------- 
(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:26.697: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:26.698: Loading IM context type 'ibus' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:26.699: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:26.699: Loading IM context type 'ibus' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.482: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.482: Loading IM context type 'ibus' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.487: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.487: Loading IM context type 'ibus' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.494: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.494: Loading IM context type 'ibus' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.500: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.500: Loading IM context type 'ibus' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.501: /lib64/libibus-1.0.so.5: undefined symbol: g_task_set_static_name

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:27.501: Loading IM context type 'ibus' failed

[... 457 lines cut ...]

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:29.703: Negative content width -1 (allocation 1, extents 1x1) while allocating gadget (node scrolledwindow, owner GtkScrolledWindow)

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:29.703: Negative content height -1 (allocation 1, extents 1x1) while allocating gadget (node scrolledwindow, owner GtkScrolledWindow)

(MeerK40t-Linux-32:109753): Gtk-CRITICAL **: 09:05:29.703: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar

(MeerK40t-Linux-32:109753): Gtk-CRITICAL **: 09:05:29.703: gtk_widget_get_preferred_width_for_height: assertion 'height >= 0' failed

(MeerK40t-Linux-32:109753): Gtk-WARNING **: 09:05:29.703: gtk_widget_size_allocate(): attempt to allocate widget with width 13 and height -25

(MeerK40t-Linux-32:109753): Gtk-CRITICAL **: 09:05:29.703: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
```

### MeerK40t Version

v0.9.3001

### MeerK40t Type

Executable

### Your Operating System

Linux

tatarize, 2023-11-23:
That's pretty spammy but basically anytime you don't give a control enough room it starts to scream like that.

jpirnay, 2023-11-23:
We will look into the cropping issue. The numerous GTK warnings unfortunately can't be suppressed nor avoided, gtk is infamous for it's attitude

jpirnay, 2023-11-23:
The cropping has been addressed ie a wxpython issue circumvented

jpagliaccio, 2023-11-26:
I am (finally) able to run meerk40t from the python git source using a conda environment (as opposed to the precompiled version). For version v0.9.3002 running on Fedora 38 with Python 3.9.18 the navigation window formatting is working nicely but I get this (missing icons). I ran from the meerk40t directly. 

[image]

jpirnay, 2023-11-26:
(just installing Fedora 39 on my vbox) This is true everywhere or just in this very window?
Hint: open console, enter ``set debug_mode True`` and then ``quit``. Restart meerk40t. You should now have a ``Debug``-menu entries under the panes menu - open Icons. Pick one from the combobox - can you see them?

jpirnay, 2023-11-26:
Hmm
[image]

[image]

Latest main...

jpagliaccio, 2023-11-30:
I did a git pull and am at version V0.9.3002 and debug is on. The Navigation icons are showing now. It was only the Navigation icons that were missing. Thanks for your help. FYI - I am still getting the warnings below ... 

At startup -------------------------

06:20:19: Debug: ClientToScreen cannot work when toplevel window is not shown
06:20:19: Debug: ClientToScreen cannot work when toplevel window is not shown
06:20:19: Debug: ClientToScreen cannot work when toplevel window is not shown
06:20:19: Debug: ClientToScreen cannot work when toplevel window is not shown

(meerk40t.py:1402909): Gtk-WARNING **: 06:20:19.899: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:20:20.019: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:20:20.019: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:20:20.044: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:20:21.827: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:20:21.827: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

Clicking on Navigation -------------------------

(meerk40t.py:1402909): Gtk-WARNING **: 06:26:10.591: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:26:10.591: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:26:18.222: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

(meerk40t.py:1402909): Gtk-WARNING **: 06:26:18.222: Negative content width -14 (allocation 20, extents 17x17) while allocating gadget (node button, owner GtkButton)

jpirnay, 2023-11-30:
From what I could establish the simplest way (not necessarily the one that you may like / need) is to reroute those messages to ``/dev/null``:
```
python3 ./meerk40t.py 2>/dev/null
```

closed 2024-01-17 by jpirnay