meerk40t/meerk40t #2327: [0.9.4b1] Simulation/burn estimates - what exactly is extra time and why is it humungous

Sophist-UK, 2023-12-06:
19mins burn time (realistic), 11secs travel time (about right), 17.5mins "extra time" for "turnaround" (WTF????!!!!).

[image]

So what exactly is extra time and how is it calculated?

Sophist-UK, 2023-12-07:
In this example, no waits - Origin, burn, Origin - nothing to account for c. 80% "extra time" being added to burn time.

tatarize, 2023-12-08:
The turn around in the raster is counted in extra time. If I otherwise count just the time it takes to go at speed it would finish significantly faster. This time is added on and calculated and timed for the decel-accel cycles. If the actual resulting time is wrong and much closer to 21 than it is to 38 then feel free to complain. But, it is the case that a bit less than about half the time spent lasering is spent turning around.

closed 2024-10-04 by jpirnay