meerk40t/meerk40t #2330: [0.8.12] Raster at DPI 1000 shows ridges

Sophist-UK, 2023-12-07:
Burning an acrylic raster at DPI 1000 I get ridges.

I tried a burn with the laser off and my finger on the Y-rail to sense the steps and it seemed to be doing 3 sweeps with a step and then 1 without.

Is this a quirk of the m2 nano?

tatarize, 2023-12-08:
The movement is very small, but it should be moving. If you're worried you can dump the LHYMICRO-GL code and look at it. Should tell you either by specifying `G001` at the start or having movements `La` or `Ra`.

Sophist-UK, 2023-12-09:
I have no idea how to "dump the lhymicro-gl code", nor how to interpret it.

I am also unsure about whether this is a MK or a controller issue (not the least because I have no idea what the control codes are for rastering on the m2 nano). The raster doesn't end up 3/4 of the size it should be in the Y direction, so overall the number of steps seems right and the missing steps are made up somehow.

So it feels either like some sort of rounding error where the y-position results in 0 steps and then 2 steps or a some sort of in-baked hardware issue.

The m2 nano is "dumb" so it seems unlikely that it is a rounding error inside the controller. It might be a rounding error inside MK, but that also seems unlikely (though in the past I have diagnosed similar rastering issues in #1921 that were rounding/offset issues in MK, so I guess this is still possible).

Looking at hardware possibilities, when I unlock the rails, and move the head by hand there are also slight but extremely regular ups and downs in resistance. I am unsure whether these are magnetic effects inside the stepper motors (the regularity is of the order of (maybe) 0.5mm (I haven't attempted to measure it accurately) but it seems too big for steps (which are c. 0.025mm), or something to do with the belt ridges but it seems too small for these (which are c. 2mm apart), or it could be something else entirely. But they feel far too regular to be something sticking and in any case it is the same on both axes which also rules out a hardware glitch IMO. 

This resistance is far too small to prevent steps, but it could be enough to use up some tiny inbuilt sloppiness that I can see no way to eliminate - so whilst the stepper moves 1 step on each sweep, i.e. three steps progressively takes up the sloppiness and the next step overcomes it and effectively moves the head 2-steps - so instead of 1-1-1-1 you get something like 1-1-0-2 (but the 2 is indistinguishable from a 1 to the naked finger - when I ran it at DPI-500 I couldn't feel the difference between a 1-step and a 2-step).

My rig is as taut and sloppiness free as I can make it, but there is some sub-mm play in the bearing on the LH Y-axis bar, and a bit more play (mainly vertical) on the RH-side where it is rollers running on a track rather than a stainless-steel bar - but when I run the raster with laser off at a slow speed (which would eliminate acceleration-related sloppiness), I still get this issue.

Since DPI 500 doesn't feel like it suffers from this irregularity (though I am not sure I could distinguish between 2-2-2-2 and 2-2-1-3, I might try this and see if it eliminates the ridges. Or I can try a cross-hatch raster and see if that is any better.

I am travelling for a week, so cannot do a lhymicro code dump until I am back, but if anyone has any suggestions in the mean-time, I would love to hear them.

Sophist-UK, 2023-12-16:
I tried burning a similar design at DPI500 and it seemed to move on every other sweep so instead of 2-2 it would be 4-0.

I have a short video of the Y stepper stepping if that will help.

Sophist-UK, 2023-12-18:
I am currently trying L2R rastering, and despite the sweeps being vertical I am still getting very fine **horizontal** lines (finer and less pronounced than with B2T rastering and very evenly spread, but perpendicular to the sweeps rather than parallel).

There seem to be c. 2-4 lines per mm - DPI 500 is step 2 or c. 0.05mm which is much smaller, so it doesn't seem likely to be a product of the stepping, nor a function of the belt ridges. 

This is on acrylic, so there is no natural grain to the material which could be causing this, but it could be something about how the material is created or it could be the PWM of the digital PSU or possibly vibrations in the Y axis.

closed 2024-10-04 by jpirnay

jpirnay, 2024-11-06:
It would be interesting to see whether this might be associated with issue #1921. The rasterplotter implementation in 0.8 is identical to the one in 0.9. This definitely had some issues with bidirectional movements if a overscan value was defined. A new implementation in PR 2715 has addressed these issues.