meerk40t/meerk40t #2332: [0.9.4b1] Suggested Image dithering improvements

Sophist-UK, 2023-12-09:
I normally work with b/w logos, so today was a bit unusual because I was working with a hi-res (1382DPI at the correct scale for the burn) greyscale drawing and trying to use the raster wizard to get the best dithering.

The first issue is performance. Just changing the dithering type on the ImageNode tab of the image properties, there was a ***MASSIVE*** difference in performance between Floyd-Steinburg (fast - c. 1s) and the rest (c. 30s). I am not sure what can be done to optimise these other algorithms, but I did note that CPU was low suggesting that Numpification could help. (Doing a newsy at a low sample size took almost an hour to complete. This sort of performance is a serious inhibitor to using MK Raster Wizards rather than e.g. Imag-R.)

One possible suggestion for making the UI faster is to apply it at the screen resolution and to the parts of the image shown - so if I am zoomed in to see what the Wizard does to individual pixels, then applying the wizard only to what I can see would make no difference. Of course if I then zoom out you would need to run it again - but you could probably calculate the full image as a background task. Similarly, if I am zoomed out to see the whole image, does it really make sense to dither it at full resolution rather than at the screen resolution (assuming that the screen resolution is lower and that the resulting displayed image looks the same.) 

The second set of issues are pure UI - on the Modification tab, if you have applied a wizard, then clearing it is done by selecting a "Set to None" value in the dropdown and clicking "Apply script", when this should probably be a "Clear" button at the bottom of the tab/window. And it might be nicer to have a small checkbox under the Apply Script button which switches between adding the selected script to the existing script or replacing the existing script with the newly selected one. (Personal preference: default is to replace, checkbox is set for replace and unset for add, so checkbox label would be "Replace existing script".) Also, when you apply a script the resulting tabs are to the left of the Modification tab when IMO they should be on the right - and when you clear a script, it switches to the ImageNode tab rather than remaining on the Modification tab.

(P.S. The names are still hopelessly useless - there is NOTHING to indicate to a user what the differences are or what they are equivalent to in other tools.)

Thirdly, having applied a script when you switch to e.g. the half-tone tab and move the sliders, the recalculation of the image happens when you have first dragged the slider to a different value not when you release the mouse button to drop the slider into a new position.  

Finally, we need a spinner (or similar) for any of these operations which may take more than a fraction of a second i.e. all image manipulation actions. (Ideally it would give an estimate / countdown of time remaining, but that is a nice to have (whilst the spinner itself is IMO reasonably essential so the user knows that processing is happening and (more importantly) when it has finished. It might also be nice to allow the user to do some sort of BREAK to stop the process and enable the UI again.

tatarize, 2023-12-15:
The problem is Pillow builds into their software Floyd-Steinburg but the other ones need to be done in pure python.

This causes an absolutely massive time difference. And since these are ultimately scanline operations (they must be performed in sequence) there does not appear to be an obvious method of doing them in numpy.

The one exception of this may actually be `newsy`, which does half-toning for the grayscale conversion rather than a typical dithering algorithm. In theory you could actually solve for all circles at the same time. You'd chop up the image parallel, find each chunk, count the black within that chunk, then do a look-up-table for the circle and output that to a different array. That in theory is numpifiable. The dithers do error forward checking where one pixel could very much end up flickering a black to white across the screen. This is, essentially defacto sequential. You cannot calculate the where the error squeezes to before you calculate the pixel values.

tatarize, 2023-12-15:
https://github.com/python-pillow/Pillow/blob/e5ee034675491d1f67410022f07256ed583cbbc5/src/libImaging/Convert.c#L1375

This C implementation for the floyd-steinberg is hard to beat, I've considered brushing off my C skills and implementing several of the other error-forward dithers.

Sophist-UK, 2023-12-15:
I found a Pillow extension Hitherdither that does other dithers, but it is written in Python not C.

But [this Youtube video](https://www.youtube.com/watch?v=Ld_cz1JwRHk) (which I haven't had time to watch) shows how to use Numba for dithering.

tatarize, 2023-12-15:
The docstring for the dither function in meerk40t:

```python
def dither(image, method="Floyd-Steinberg"):
    """
    This function and the associated _DIFFUSION_MAPS taken from hitherdither. MIT License.
    :copyright: 2016-2017 by hbldh <henrik.blidh@nedomkull.com>
    https://github.com/hbldh/hitherdither
    """
```

It's not really a full library but did get a nice set of diffusion maps. 
hbldh/hitherdither#9

tatarize, 2023-12-15:
The suggestion to use numba might actually be workable. I have fallback code already and could probably hammer out the right stuff pretty quickly. So if you had numba installed it could maybe run faster than if you didn't. Which is sort of common to have around.

tatarize, 2023-12-16:
Okay. Implemented. With numba installed to compile them they go from around 200 seconds to  ~0.21 - 0.09s. The big deviation there is the first execution requires compiling the LLVM compiles.

tatarize, 2023-12-16:
Ironically this is kind of a mismash of your two suggestions there. It's numba accelerated hither-dither.

Sophist-UK, 2023-12-16:
So - can you include Numba in the executables?

tatarize, 2023-12-16:
That is an open question. I would assume so, it would just work under the hood there and it seems like there was support for our supported metrics, But, it needs to properly either throw an import error (to fallback to normal), and not do anything else that's weird. Which means some checking.