meerk40t/meerk40t #2348: [New requirement] Opposite of `Break Subpaths" needed

Sophist-UK, 2023-12-16:
I can break apart a combined path, but I cannot then recombine them (e.g. if I want odd-even to apply).

Can we please have a right click function so that when 2 or more paths (or path-like objects) are selected we can then combine them into a single path.

tatarize, 2023-12-16:
Merge the group, should work. I recently did this accoringly.

Sophist-UK, 2023-12-16:
I think you can only merge a group. To use this you would have to Group the selected items first and then merge - assuming this works (I haven't tested it), then it is not intuitive or efficient and a combine paths command would be helpful.

tatarize, 2023-12-17:
Break Subpaths breaks into a group. It's probably absolutely correct to do things that way and it works fine. You create a group of the subpaths, sort through them do your work and pack them back into a single path.

Sophist-UK, 2023-12-17:
That assumes that you want to put them back into a single path. But if you have text-to-path containing a phrase and you want to break it into separate letters and you need odd-even processing to make e.g. the centre of an "o" white, then after you break it apart into a new group, you then need to combine two paths and **not** every path in the original group.

tatarize, 2023-12-17:
I hear inkscape recently changed the way their first text-to-path things worked in that it grouped things by the word, by the letter, then by the contour. Since you reassembling individual diacritics from a pile of contours is both really hard and not what the user should probably want.

closed 2024-01-17 by jpirnay