meerk40t/meerk40t #2349: [Suggestion] Improved tree highlighting

Sophist-UK, 2023-12-16:
In order to show the user when in the tree node(s) selected in the scene are located, I would like to propose the following enhancement to the tree:

1. A new highlighting classification of "related" shown with a faint grey background. Suggested that "emphasised" items are shown in a faint blue background (cf. "selected" items being dark blue).

2. When items are selected or emphasised, "related" items are highlighted - with the following items being related:
    * Containing nodes e.g. Group/file/op nodes containing selected or emphasised items
    * Matching elements in other major branches of the tree (and their containing nodes)

So if I select an element on the scene, then in addition to selecting the actual nodes in the Elements part of the tree and emphasising any child nodes, you also mark any parent nodes, and any corresponding nodes in Ops and their parent nodes as "related". I can then immediately see which (closed) File/Group nodes they are part of, even if those nodes are closed, and I can see which ops they are part of.

tatarize, 2023-12-17:
```python
        """
        If any operation is selected, all sub-operations are highlighted.
        If any element is emphasized, all references are highlighted.
        If any element is emphasized, all operations a references to that element are 'targeted'.
        """
```

Currently I believe this is what the current system looks like.

Whereas the enhancements are:
```python
    def set_enhancements(self, node):
        """
        Node in the tree is drawn special based on nodes current setting.
        @param node:
        @return:
        """
        tree = self.wxtree
        node_item = node._item
        if node_item is None:
            return
        if self._freeze or self.context.elements.suppress_updates:
            return
        tree.SetItemBackgroundColour(node_item, None)
        try:
            if node.highlighted:
                tree.SetItemBackgroundColour(node_item, wx.LIGHT_GREY)
            elif node.emphasized:
                tree.SetItemBackgroundColour(node_item, wx.Colour(0x80A0A0))
            elif node.targeted:
                tree.SetItemBackgroundColour(node_item, wx.Colour(0xA080A0))
        except AttributeError:
            pass
```

The reference of an emphasized node is highlighted.  I don't think anything actually ever targets a node.

Sophist-UK, 2024-10-04:
This has AFAIK not been completed.

I appreciate that this may be low priority, and may never get done, but closing it means it will never get done.

This feels like a "fuck you" moment. I make a reasonable suggestion for a UI improvement and it isn't rejected due to a difference of opinion or because it is impossible to do. Instead it is simply closed in order to reduce the number of open issues. _This is just like a call centre operative getting their queue down to zero by picking up and then dropping calls in quick succession._ If this is the respect given to people donating their time to make suggestion on how to improve MK still further then I won't bother in future!!

jpirnay, 2024-10-05:
Reopened to allow someone to pick it up at a later time, if there's a wider use case for it.