meerk40t/meerk40t #2352: [Bug]: Lightburn integration stopped working on every computer

AtrujanoX, 2023-12-18:
### Summary Description

I had lightburn with ruidacontrol, tonight at 00:00 suddenly the system stopped working and the next error is coming out.

### Additional Details

Maybe is this an expired certificate issue?

### Crash logs

```shell
MeerK40t crash log. Version: 0.9.3001 on Windows: Python 3.8.10: AMD64 - wxPython: 4.2.1 msw (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "meerk40t\ruida\rdjob.py", line 289, in execute
  File "meerk40t\ruida\rdjob.py", line 697, in process
IndexError: list index out of range

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "meerk40t\kernel\kernel.py", line 1740, in run
  File "meerk40t\core\spoolers.py", line 543, in run
  File "meerk40t\ruida\rdjob.py", line 291, in execute
meerk40t.ruida.exceptions.RuidaCommandError: Could not process Ruida buffer, [[209, 24], [175], [255], [255], [255], [211], [137], [255], [255], [255], [147], [129, 24], [174], [255], [255], [252], [235], [255], [255], [253], [208], [209, 24], [251], [255], [254]] with magic: 17

During handling of the above exception, another exception occurred:

wx._core.wxAssertionError: C++ assertion "wxThread::IsMain()" failed at ..\..\src\msw\thread.cpp(1432) in wxMutexGuiLeaveOrEnter(): only main thread may call wxMutexGuiLeaveOrEnter()!

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.IdleEvent'> returned a result with an error set
```

### MeerK40t Version

0.9.3001

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-12-18:
I wouldn't assume certification because there are not certs. The protocol is very lightly swizzled and sent plain over UDP. 

It looks a bit like the magic number was switched and switched wrongly. Your given packet is set to magic number 17 but appears to be off by one and really a magic number 18 packet. 

Though really I don't see exactly why that would crash wx at all.

AtrujanoX, 2023-12-18:
this is the error from win event viewer, 
- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" /> 
  <EventID>1000</EventID> 
  <Version>0</Version> 
  <Level>2</Level> 
  <Task>100</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8000000000000000</Keywords> 
  <TimeCreated SystemTime="2023-12-18T06:24:20.8596239Z" /> 
  <EventRecordID>7551</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="16204" ThreadID="11760" /> 
  <Channel>Application</Channel> 
  <Computer>nova</Computer> 
  <Security UserID="S-1-5-21-1544198327-415450045-3774987928-1001" /> 
  </System>
- <EventData>
  <Data Name="AppName">MeerK40t.exe</Data> 
  <Data Name="AppVersion">0.9.3001.0</Data> 
  <Data Name="AppTimeStamp">6558ee43</Data> 
  <Data Name="ModuleName">wxmsw32u_core_vc140.dll</Data> 
  <Data Name="ModuleVersion">3.2.2.1</Data> 
  <Data Name="ModuleTimeStamp">647fe3a0</Data> 
  <Data Name="ExceptionCode">c0000005</Data> 
  <Data Name="FaultingOffset">00247678</Data> 
  <Data Name="ProcessId">0x38c8</Data> 
  <Data Name="ProcessCreationTime">0x1da3156cac0a948</Data> 
  <Data Name="AppPath">C:\Users\atruj\Downloads\MeerK40t.exe</Data> 
  <Data Name="ModulePath">C:\Users\atruj\AppData\Local\Temp\_MEI153282\wx\wxmsw32u_core_vc140.dll</Data> 
  <Data Name="IntegratorReportId">95277de7-08d5-49d3-892c-4bfc27ceca7d</Data> 
  <Data Name="PackageFullName" /> 
  <Data Name="PackageRelativeAppId" /> 
  </EventData>
  </Event>

AtrujanoX, 2023-12-18:
sorry I rushed and forgot to mention the crash happens whenever a drawing is sent from lightburn to mrk4t via the ruidacontrol socket, firstly I thought it had something to do with line and fill settings, but then I switched to my main pc and realized the crash started to happen on fresh installs too

tatarize, 2023-12-18:
Really this happens if something sent a UDP packet to the device that wasn't valid Ruida code. In this case the data sent seems like it's gibberish. Undoing the swizzle gives us:

`b'\xc1\n\xbf\xef\xef\xef\xc3\x99\xef\xef\xef\x83\x91\n?\xef\xefm\xfb\xef\xef\xedA\xc1\n\xeb\xefo'`

Magic number 0x11 (17) is fairly common. 

If we used `ee` for the decoding we'd get the commands:

b'.\xe7P\x00\x00\x00,v\x00\x00\x00l~\xe7Q\x00\x00\x03\x14\x00\x00\x02/.\xe7\x04\x00\x01'

\xe7 (P\x00\x00\x00,v\x00\x00\x00l~)
\xe7 (Q\x00\x00\x03\x14\x00\x00\x02/.)
\xe7 (\x04\x00\x01...)'

The command E7 0x50 is `Document Mid Point` this is does require 10 payload bytes.
The command E7 0x51 is `Document Max Point` this also requires 10 payload bytes.
The E7 0x04 is `Process Repeat`.

In typical execution these commands would appear together. It appears that somehow the magic number being used by Lightburn is 0xEE and somehow it's getting encoded as 0x11.

tatarize, 2023-12-18:
I think what's happening is that `lightburn` tends to send a bunch of queries to see what swizzle the device uses. Typically this is query for the CardID and the sends some variation of `\xda\x00\x05\x7E` which a real ruida machine would reply to just the one correct magic key. MeerK40t just replies to all of them, and uses the query to determine what key it should use. So regardless what talks to it, sending any of the CardID queries will cause Meerk40t's magic number to change. I think maybe due to the UDP that system can get screwed up and lightburn reacts to one of the other queries. So they end up with different magic numbers.

Parsing your command seems to show that meerk40t was reading magic number 0x11 but lightburn was sending magic number 0xee data.

I think maybe this stuff needs an override, to force the value. Or a secondary check when the main block of data is sent.

I'm not sure why that crash would mess up anything other than failing to send and giving you the crash log accordingly. It may actually be fixed by closing down and reopening Lightburn since it's actually a desync between the two programs.

AtrujanoX, 2023-12-18:
I discovered the frame and home operations work properly, also restarting Lightburn does not work, I've tried starting each program in different order, none of them order combinations are working, I just downloaded the python version and put a print on the array to debug if anything is useful, I really don't understand the code completely yet

```
meerk40t -e ruidacontrol -c
←[0m[03:44:52] ←[34m←[1mruidacontrol←[0m
←[0m[03:44:52]     ←[31mWARNING: ←[34mNon-horizontal rasters may not work well.←[0m←[0m
←[0m[03:44:52]     ←[31mWARNING: ←[34mCuts are expected below 80mm/s. Rasters above that speed.←[0m←[0m
[03:44:52]     Ruida Data Server opened on port 50200.
[03:44:52]     Ruida Jog Server opened on port 50207.
>>[239, 25]
[254, 15, 14]
[253]
[255, 39]
[255, 24]
[249]
[255]
[255]
[255]
[255]
[255]
[255]
[255]
[255]
[255]
[255, 24]
[199]
[255, 24]
[252]
[255]
[255]
[248]
[215]
[255]
[255]
[252]
[210]
[167, 24]
[248]
[255]
[255]
[240]
[136]
MeerK40t crash log. Version: 0.9.3001 pkg on Windows: Python 3.10.6: AMD64 - wxPython: 4.2.1 msw (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 290, in execute
    self.process(array)
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 417, in process
    x = abscoord(array[1:6]) * self.scale
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 79, in abscoord
    return decode32(data)
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 55, in decode32
    return signed32(decodeu35(data))
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 60, in decodeu35
    (data[0] & 0x7F) << 28
IndexError: list index out of range

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\meerk40t\kernel\kernel.py", line 1740, in run
    func_result = func(*args)
  File "C:\Python310\lib\site-packages\meerk40t\core\spoolers.py", line 543, in run
    fully_executed = program.execute(self.driver)
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 292, in execute
    raise RuidaCommandError(
meerk40t.ruida.exceptions.RuidaCommandError: Could not process Ruida buffer, [[247], [255], [255], [245], [145], [175], [255], [255], [248], [215], [255], [255], [252], [210], [167, 24], [174], [255], [255], [240], [136], [247], [255], [255], [245], [145]] with magic: 17
```

and this is the output from a framing

`
>>[201, 2, 0, 0, 0, 0, 0]
[198, 1, 0, 0]
[198, 2, 0, 0]
[198, 33, 0, 0]
[198, 34, 0, 0]
[201, 2, 0, 0, 6, 13, 32]
[201, 2, 0, 0, 0, 0, 0]
`
hope it helps

tatarize, 2023-12-18:
`meerk40t -ce "ruidacontrol -v"`

Will open it up and give you the readout of exactly what's happening.

AtrujanoX, 2023-12-18:
Updating lightburn helped, I just installed the newest version 1.4.04, which I didnt want because of debugging but I just want it to work already, i didn't check for new versions and seems now to work properly, for the record, 1.4.00 is not working correctly from today somehow, but yesterday was perfectly, ¯\_(ツ)_/¯. thanks for helping me <3

tatarize, 2023-12-18:
Should launch the server but in verbose mode. And basically translate the communications for you. It's a bit spammy but might be fine if you're using it in textbased mode. The somewhat relevant data is when it sets the magic number to 0x11

tatarize, 2023-12-18:
I'll give it some additional thought, since really you should be able to set a magic number and force that magic number to whatever you need it to be. So this could at least get a bit better sorted. I had tried but I was trying the latest lightburn and it was working fine for me.

tatarize, 2023-12-18:
It still helps a lot, since it clearly establishes that this error is because lightburn and meerk40t settled on different magic numbers. Though I haven't seen it since I made meerk40t just accept every version of `da00057e` and use that to set the magic number.

AtrujanoX, 2023-12-18:
Below is the log from the verbose execution with 1.4.00 (from the laptop i didnt update yet), I really can't find out what just made it behave different from today, as you said, it's just an UDP socket and suddenly started to send different data.
```
meerk40t -ce "ruidacontrol -v"
←[0m[04:09:05] ←[34m←[1mruidacontrol -v←[0m
←[0m[04:09:05]     ←[31mWARNING: ←[34mNon-horizontal rasters may not work well.←[0m←[0m
←[0m[04:09:05]     ←[31mWARNING: ←[34mCuts are expected below 80mm/s. Rasters above that speed.←[0m←[0m
[04:09:05]     Ruida Data Server opened on port 50200.
[04:09:05]     Ruida Jog Server opened on port 50207.
>>[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000004     (Get 00 04 (mem: 0004) (IOEnable))
[04:09:30]     <-- da0100040000000000   (Respond 00 04 (mem: 0004) (IOEnable) = 0 (0x00000000))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000026     (Get 00 26 (mem: 0026) (Axis Range 1, Get Frame X))
[04:09:30]     <-- da0100260000134400   (Respond 00 26 (mem: 0026) (Axis Range 1, Get Frame X) = 320000 (0x0004e200))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000036     (Get 00 36 (mem: 0036) (Axis Range 2, Get Frame Y))
[04:09:30]     <-- da01003600000d3660   (Respond 00 36 (mem: 0036) (Axis Range 2, Get Frame Y) = 220000 (0x00035b60))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000021     (Get 00 21 (mem: 0021) (Axis Precision 1))
[04:09:30]     <-- da0100210000000000   (Respond 00 21 (mem: 0021) (Axis Precision 1) = 0 (0x00000000))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000031     (Get 00 31 (mem: 0031) (Axis Precision 2))
[04:09:30]     <-- da0100310000000000   (Respond 00 31 (mem: 0031) (Axis Precision 2) = 0 (0x00000000))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000010     (Get 00 10 (mem: 0010) (System Control Mode))
[04:09:30]     <-- da0100100000000000   (Respond 00 10 (mem: 0010) (System Control Mode) = 0 (0x00000000))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00001e     (Get 00 1e (mem: 001e) (Auto Type Space))
[04:09:30]     <-- da01001e0000000000   (Respond 00 1e (mem: 001e) (Auto Type Space) = 0 (0x00000000))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000004     (Get 00 04 (mem: 0004) (IOEnable))
[04:09:30]     <-- da0100040000000000   (Respond 00 04 (mem: 0004) (IOEnable) = 0 (0x00000000))
[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000004     (Get 00 04 (mem: 0004) (IOEnable))
[04:09:30]     <-- da0100040000000000   (Respond 00 04 (mem: 0004) (IOEnable) = 0 (0x00000000))
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da000226     (Get 02 26 (mem: 0126) (User Para 1))
[04:09:30]     <-- da0102260000000000   (Respond 02 26 (mem: 0126) (User Para 1) = 0 (0x00000000))
[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:30]     Setting magic to 0x88
[04:09:30]     <-- cc   (Checksum match)
[04:09:30]     --> da00057e     (Get 05 7e (mem: 02fe) (Card ID))
[04:09:30]     <-- da01057e0628014a00   (Respond 05 7e (mem: 02fe) (Card ID) = 1694524672 (0x65006500))
[04:09:31]     <-- cc   (Checksum match)
[04:09:31]     --> da000004     (Get 00 04 (mem: 0004) (IOEnable))
[04:09:31]     <-- da0100040000000000   (Respond 00 04 (mem: 0004) (IOEnable) = 0 (0x00000000))
[04:09:31]     <-- cc   (Checksum match)
[... 259 lines cut ...]
[04:09:41]     -**-> f9 (Unknown Command!)
[242]
MeerK40t crash log. Version: 0.9.3001 pkg on Windows: Python 3.10.6: AMD64 - wxPython: 4.2.1 msw (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 290, in execute
    self.process(array)
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 904, in process
    if array[1] == 0x00:
IndexError: list index out of range

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\meerk40t\kernel\kernel.py", line 1740, in run
[... cut ...]
  File "C:\Python310\lib\site-packages\meerk40t\ruida\rdjob.py", line 292, in execute
    raise RuidaCommandError(
meerk40t.ruida.exceptions.RuidaCommandError: Could not process Ruida buffer, [[223, 57], [255], [230], [178, 57], [255], [230], [178, 57], [190], [255], [230], [178, 57], [189], [255], [230], [178, 53], [249], [255], [255], [255], [255], [255], [255, 53], [190], [255], [255, 24]] with magic: 17

[04:09:45]     <-- cc   (Checksum match)
[04:09:45]     NOT A COMMAND: 24
[04:09:45]     Process Failure: 18
```

AtrujanoX, 2023-12-18:
[image]
The rectangle I used for testing

tatarize, 2023-12-18:
The communications are going fine, then suddenly it gets an entirely wrong value or something. But, there was no indication that value shifted. But, clearly from the log there is a desync between 0x88 and 0x11 being used. The crash says it failed with magic 17 but the last bit in communications had magic 0x88 being used.

tatarize, 2023-12-18:
Okay. Found it. Okay, found the reason for the wrong value being given there, though it still doesn't explain the sudden communication garbling. I'm going to give 1.4.0 a try.

tatarize, 2023-12-18:
1.4.00 worked fine too. Cannot reproduce.

tatarize, 2023-12-21:
I believe I may have experienced this but it got better fairly quickly.

closed 2024-03-19 by jpirnay