meerk40t/meerk40t #2358: [Bug]: locks up or takes hours when using slightly complex jobs

scarr999, 2023-12-21:
### Summary Description

I have tried several jobs that are A4 and have more than a simple job on them and Meerk40t goes extremely slow, 3.7Ghz CPU, 64GB Ram, M.2 drive (so not the slowest of PC's) 

I have added the SVG, the SVG has four layers as I tried to section it and output these smaller sections to separate files, hopefully take the load off Meerk40t but it's still too slow. ideally, I'd like to just have one layer and engrave and cut in one pass, not 4 passes (Engrave 1,2 Cut 1,2)

### Additional Details

[Velociraptor single sheet.zip](https://github.com/meerk40t/meerk40t/files/13737793/Velociraptor.single.sheet.zip)

### MeerK40t Version

0.9.3010

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-12-21:
Turn off `Optimization` it seems like that's going really slowly.

You could do the next bit of optimizations that's coming down the pike to speed that up.

tatarize, 2023-12-21:
You can also right-click the main group and `Merge Items` That'll create a single big item to rule them all. Seem to optimize rather fast. 

Also, it's a small grid of hearts what the heck takes 6 megs?

tatarize, 2023-12-21:
Tried without merging it seemed to take like 2-3 seconds. Ah, I have different settings you have many of these passes and it can't just do the same heart each time? So it's actually optimizing 1000 x 11 x 10 different hearts which aren't allowed to simply do the first heart 1000x in a row.

tatarize, 2023-12-21:
Make four operations. Likely merge the giant set of 110 hearts into 1 path. Then add it to each of those operations.

I can't seem to get this to go slowly. Am I missing something?

[image]

scarr999, 2023-12-21:
It's not just Hearts, it has layers, turn them all on in the SVG and there is a lot more. I made the layers in an attempt to help meerk40t speed, maybe just running layers off does nothing to help?

tatarize, 2023-12-21:
Seems like it should help some. Since it has fewer objects to deal with. This one seems to go fairly reasonably in most cases. Usually the raw number of objects causes some slowdowns. And you're better off merging some paths together to have fewer nodes to deal with. Which you can sort of do in meerk40t.

scarr999, 2023-12-21:
OK we are going to get mixed up as we are both doing things to try to fix it :-) 

When I load the SVG (Optimize off) and only showing "engrave 1" layer, I first clear all operations, I then select all of the "engrave 1" items and this alone takes a while, then once it is finished selecting them I try to "assign operation" to the selected group and it takes forever.

tatarize, 2023-12-21:
Mostly though if you turn off optimizations it should go a good clip faster, though it might lose some time in moving things around.

tatarize, 2023-12-21:
Drag and drop the entire group into Engrave 1.

tatarize, 2023-12-21:
Okay, turned all the layers on. So I can experience the full set of issues. The initial loading is decidedly going a lot slower now.

tatarize, 2023-12-21:
I guess loading 1 single copy and then doing `element* grid 10 11` or whatever that command is would let you just create your own version of sorts. There's not too much to do to speed up the load time of a file like that.

Do you want to run this design in series or in parallel. Each card fully finished before starting the next card, or do all the first elements of the card and then all the second elements of the card and repeat?

scarr999, 2023-12-21:
So, I have just "engrave 1" on, to load the file takes 29 seconds, to select just "engrave 1" items by dragging a box around them 1min 29 seconds. 

I was then right-clicking and looking for "merge items" but I cannot see this option, did you mean "group" not merge?

tatarize, 2023-12-21:
Okay, for some reason it thinks this file should have raster, cut, raster, cut for a very long time.

No, if the items are in a group you can `Merge` that group. 

[image]

scarr999, 2023-12-21:
> I guess loading 1 single copy and then doing `element* grid 10 11` or whatever that command is would let you just create your own version of sorts. There's not too much to do to speed up the load [...]

Each card finished before next

tatarize, 2023-12-21:
I think our current solution for this is a placement grid. Basically just let run 1 job in the right order in all 110 locations.

tatarize, 2023-12-21:
Running each card in order is a bit different than merely running the Engrave, Engrave, Cut, Cut. Since really you want to run 110 Engraves, 110 Engraves, 110 Cuts, 110 Cuts. And do each followed by the next.

tatarize, 2023-12-21:
Merging would not help with that since they would each be their own object and what you want is 1 set of objects run 110 times.

scarr999, 2023-12-21:
> Running each card in order is a bit different than merely running the Engrave, Engrave, Cut, Cut. Since really you want to run 110 Engraves, 110 Engraves, 110 Cuts, 110 Cuts. And do each followed by the next.

I simply want to do a A4 sheet, 

First, engrave everything (all engrave as one job) 
Then the hearts as one cut
Finally the outline as one cut. 

I am not sure why meerk40t is making all the engrave's cut's. unless it's something I did in inkscape?

tatarize, 2023-12-21:
It might be a feature, since you labeled it `Cut` or something. I forget how it decides that based on label there.

scarr999, 2023-12-21:
Thanks for all your help, know you have looked at a lot of things what is your best suggestion to make this work?

tatarize, 2023-12-21:
Single cell of the design run 110 times across the field.

tatarize, 2023-12-21:
[image]

I think I may have done the values wrong.

[image]

tatarize, 2023-12-21:
Depending a bit on your bed size. That'd be nearly right. Though I have 12 additional rows down and meant to do 10x11.

Basically create 1 placement in the corner.
Create placements horizontally like 25mm apart.
Select all the placements that are horizontal and right-click and Create placements Vertically, like 20 mm apart.

tatarize, 2023-12-21:
[image]

This is basically a forced move because if you want to do the full card with each step in the correct order, you need to use placements. Or you're forced to design 11 x 10 x 4 different operations. And making sure each of them is in the right order.

So here you see the simulation partially done, and it did my top row first, then it did all the additional vertical rows. It's basically the only solution here because even if you could tame the sheer number of objects, we're dealing with a downright crazy amount of work to give them each their own properly parallel bit of code.

scarr999, 2023-12-21:
> Single cell of the design run 110 times across the field.

Thanks for all your help, will try this and it makes sense.

tatarize, 2023-12-21:
It's entirely true that this is really hard, and the *solution* here is kind of hard to figure out, and really it *should* be a lot easier.

tatarize, 2023-12-21:
It's a somewhat sane amount of time to simulate the design though. And hopefully in 0.10.x stuff that issue will go away too since we won't need to duplicate the code for each of these positions. Much more effective than dealing with that many objects but still sub-optimal.

tatarize, 2023-12-21:
Though I did do the simulation without Optimize, which is probably preferred though maybe without `Merge Operations` since if it can smash the cards together between operations it might actually violate the whole `1 card` then `next card`... requirement there.

scarr999, 2023-12-21:
> It's a somewhat sane amount of time to simulate the design though. And hopefully in 0.10.x stuff that issue will go away too since we won't need to duplicate the code for each of these positions. Much more effective than dealing with that many objects but still sub-optimal.

Sorry to be a newbie, but where is the option to run 110 times in a grid like fashion?

tatarize, 2023-12-21:
Placements. You actually need to make a row of Horizontal Placements from 1 single start placement and then a row of Vertical placements from that row of Horizontal placements.

The SVG I posted can be right-click downloaded and will have the placements in them already. Run that without optimize and it should be fairly reasonable.

tatarize, 2023-12-21:
https://www.youtube.com/watch?v=6WT2RD69j3k

I think I go over making a makeshift grid. Really it needs a grid placement thing but I'll toss that in feature suggestions on discord.

jpirnay, 2024-03-19:
Not a bug, optimisation speed needs to be addressed. Specific case is better done with placements

closed 2024-03-19 by jpirnay