meerk40t/meerk40t #2383: [0.8.12 BUG] Edge case: Burn inner first is missing a few burns where end nodes are coincident with outer path nodes

Sophist-UK, 2024-01-01:
In inkscape the inner nodes were explicitly snapped to edge nodes, yet the edge was burned first.

I suspect that this is either a floating point rounding error or an issue with the way the algorithm approximates the outer or inner paths- the inside check needs to allow inner paths to be fractionally (i.e. say 1/2 the width of a laser beam) outside the outer path.

jpirnay, 2024-10-16:
Couldn't establish a case where that became evident. Do you have a culprit file to be showed/shared?

Sophist-UK, 2024-10-16:
This is (literally) an edge case, perhaps due to comparisons between floating point coordinates, probably in the same code referred to in the issue about images being inside [#2192](https://github.com/meerk40t/meerk40t/issues/2192#issuecomment-2395584766) where I said:

> P.P.S. There also seems to be a potential bug around use of the rounding tolerance. In the raster-specific tests, the tolerance is applied the opposite way from the following tests and I cannot work out why.

If the tolerance was applied the wrong way around, then it would likely have this effect.

jpirnay, 2024-10-17:
Fine, I created an edge case myself. But - and that's a bit but - the current scan beam implementation in geomstr as well as the old VectorMonotizer do not support at all (scan beam) / incompletely (VectorMonotizer) tolerances. So this needs to go to the backlog for now. But it's unfortunately not a simple arithmetic error.

[image]

jpirnay, 2024-10-17:
It took me some time but I think this much better: [Commit](https://github.com/meerk40t/meerk40t/commit/d0bc84078a5812e9758352fa29a7ca6f76e0bd1e)
- The early bounding box check was faulty
- The is_inner check did not recognize the tolerance

[image]

closed 2024-10-17 by jpirnay