meerk40t/meerk40t #2385: [Suggestion]: How can I help translate the software into my language?

Bpergher, 2024-01-02:
### Summary Description

I would like to collaborate in preparing the translation of the software into my language (Brazilian Portuguese), but I am new to Git and I don't really know how to add my translation to the project.
I would appreciate it if someone could guide me.

### MeerK40t Version

0.9.3

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2024-01-03:
We often do take translations in the translations discord channel.
https://discord.gg/GUXMXQD2BD

Also, because of the absolutely huge amount of work we often try to make sure things are as easy on you as we can muster. This typically means we'll do a fast pass of the translation with ChatGPT and let you mostly fix those translations since really it's just an amazingly large amount of things that would otherwise need translations and it's a bit easy to go from already typed text to fixing that text.

If you wanted to do a direct PR you'd fork the project, edit the file in your own fork, and then start a PR to merge your translated version of the text to the project. We should already have a very old entry for pt_BR it just doesn't have any real work done there.

tatarize, 2024-01-03:
The default file is located here: https://github.com/meerk40t/meerk40t/blob/pt-br-translation/locale/pt_BR/LC_MESSAGES/meerk40t.po

It'll get some default translations. But, basically all the `msgid` in English need to be translated to correct versions of the `msgstr` in pt_BR. Additionally there's some special rules with things like menu options with `&` characters largely these need translated keystrokes often. Sometimes Ctrl+S for save is wrong in a region. Also anything in brackets `{speed}`  sort of thing need to remain in English since that's a formatting string.

tatarize, 2024-01-03:
I did the first 7k lines with AI and you can see them on the above referenced file. 

There's like 15k though a few thousand lines are orphaned translations.

[image]

tatarize, 2024-01-04:
I finished the entries with AI. Keep in mind these are not always fantastic they are better than nothing and saves a *considerable* amount of typing if they are close enough. But they need a checker who speaks both languages to figure out where the AI goes wrong.

https://github.com/meerk40t/meerk40t/blob/main/locale/pt_BR/LC_MESSAGES/meerk40t.po

[image]

I think it translating `_TREE_` stuff went weird. But I borrowed the translation from jpirnay's German version so this stuff should have all the needed work to translate everything.

And it does a bunch of really hard stuff like help for console commands:

[image]

The reason for the pre-translate AI stuff is that even with the best intentions, there's just so much stuff to translate that people easily get overwhelmed. This hopefully produces pretty good results with a non-overwhelming workload.

Bpergher, 2024-01-05:
Thank you very much for the information and tips.
It really is exhausting work to translate everything manually. I even started translating with a lot of desire to finish, but little by little I realized that it was really a lot of work.
I'm glad you've already managed to complete this stage of translating Brazilian Portuguese with AI.
I've already taken a look at the result and it looks very good. I will review the translations as soon as possible.

Thanks again for the work. It is excellent!

closed 2024-02-06 by jpirnay