meerk40t/meerk40t #2391: [Bug]:  ruidaemulator and ruidadesigner commands not working

g1r1, 2024-01-04:
### Summary Description

Are the commands ruidaemulator and ruidadesign no longer available in 0.9.2? I remember using it earlier.  

Now, I get the following error.
"ruidaemulator is not a registered command in this context: Base"

Not sure if this is a bug?

Also, is there a way to have some similar functionality in grblcontrol? 

### MeerK40t Version

0.9.2000

### MeerK40t Type

Executable

### Your Operating System

Mac (M1/Bigsur)

g1r1, 2024-01-04:
Never mind. I found it simulates directly. Haven't used Meerk40t for a while I guess :)

closed 2024-01-04 by g1r1

tatarize, 2024-01-07:
Yeah, these were mostly removed.