meerk40t/meerk40t #2393: [0.9.4b2] If you have fixed all the UI scaling issues, then another beta for testing please...

Sophist-UK, 2024-01-04:
I really want to start to use 0.9 in production, but I can't until there is a version whose UI is usable on my Surface 3 Pro...

jpirnay, 2024-01-04:
The remaining scaling "issues" have been addressed with PR 2394. There's a very ugly bug in wxpython that does not properly set bitmaps with a defined size on systems with a display user scale of more than 150% (at least that's the boundary on my system). Out of a sudden bitmaps are used and displayed enlarged....
I had to circumvent this by scaling some images down to counterbalance that unwanted blowup to achieve a defined pixel size. So on systems with scaling beyond that boundary images actually look again a bit more pixely than you would expect. Even with the current beta you can achieve correct sizing if you would reduce the user scale below the 150% threshold.

Sophist-UK, 2024-01-04:
Then can I please have a new executable to test with?

tatarize, 2024-01-05:
I'll push up the beta stuff. Gotta do some bug fixes and a version bump. And the numba dither stuff needs to be tested in .exe form to make sure it's fine.

tatarize, 2024-01-10:
https://github.com/meerk40t/meerk40t/releases/tag/0.9.3020

Sophist-UK, 2024-01-10:
I'll try the new beta next time I am burning.

closed 2024-01-14 by tatarize