meerk40t/meerk40t #2408: [Bug]: Text gets clipped

kostex, 2024-01-15:
### Summary Description

Today after I've entered a long text (36chars) I've noticed that the last character did not print.
Looking at the preview/simulate, it was missing that character also.
It seems that with different kinds of fonts, the right gets clipped (independent of long strings I've noticed) a little bit. (not complete characters but part of them)
In my case the last letter was an i which got clipped completely.
(Franklin Gothic Medium, Arial Black are examples of where clipping occurs)

### MeerK40t Version

v0.9.4 B2-2

### MeerK40t Type

Executable

### Your Operating System

Windows

kostex, 2024-01-15:
Hmm, bummer,
now I've noticed that the top of a text can get clipped too..
[image]

jpirnay, 2024-01-17:
Acknowledged, this is an issue. Unfortunately it seems to be a hard nut to crack... I recommend you use vector text until we figured it out

tatarize, 2024-01-18:
Behind the scene the measuring of the size of the text within the gui toolkit sucks. It was so bad we ended up drawing on an image behind the scenes and then measuring the size of that image. This is because the behind the scenes render of the text is insufficient to cover the entire area and thus can't manage to figure out the right size for the would-be result.

jpirnay, 2024-01-23:
We have an experimental fix, ie rewrite of the text drawing algorithm in place, that no longer clips the text - PR #2411 
This needs to undergo thorough testing and code reviews before it can become part of the main codebase

kostex, 2024-01-23:
When editing/typing text and (accidentaly) deleting all characters, the application crashes.

kostex, 2024-01-23:
I hope I don't annoy too much ;-) but I've found a font, eager to give you a migraine regarding boundary calculation...
It should say pity.. 
GoogleFonts/AlexBrush-Regular
[image]

jpirnay, 2024-01-23:
You recognize this is an 'awful' font, as it breaks all given character boundaries and makes any calculations about textsize more or less useless. Still a fix is in place...

kostex, 2024-01-24:
It doesn't bother me at all, but in the properties window of the text, the preview is still clipped...

jpirnay, 2024-01-24:
> It doesn't bother me at all, but in the properties window of the text, the preview is still clipped...

I circumvented this, but this due to the poorly designed font with inappropriate kerning information.

jpirnay, 2024-01-24:
> Could I persuade you to make the line_gap variable into a user changeable setting?
> Would be cool with all those silly fonts out there not giving a cr*p about line height..

Done

kostex, 2024-01-24:
> > It doesn't bother me at all, but in the properties window of the text, the preview is still clipped...
> 
> I circumvented this, but this due to the poorly designed font with inappropriate kerning information.

Could you have a look again please? The preview now only shows the last line of text when multiline usage.
Anyway.. great how fast you respond man! Kudo's! Very nice indeed!

jpirnay, 2024-01-24:
The preview area will not resize - so it will for bigger / multilined inputs provide a partial view only. The best preview is actually on the scene.... :-)

jpirnay, 2024-01-24:
Nonsense - it's a bug, sorry ...

jpirnay, 2024-01-24:
One would not disagree ;-)

kostex, 2024-01-26:
What is the policy regarding merging this branch into main?
Thanks

kostex, 2024-01-26:
oh ok!.. I have a private local branch with this and main merged. I am/was just interested in the policy.. I know now.. and in my case I use inkscape separate from mk.. so no probs for me ;-)
Thanks again

tatarize, 2024-01-27:
Policy wise, if you have a branch you'd like merged you open a PR. We'll review it and explain if there's any issues.