meerk40t/meerk40t #2410: [Bug]: Toggle (Primary) Grid does not work until zoom in/out

kostex, 2024-01-17:
### Summary Description

When toggle show (primary) grid, the UI reflects correctly but I need to zoom in/out a bit to actually reflect the new grid setting.
i.e. turning grid off still snaps the objects to grid positions unless I zoom in/out a bit (maybe other options trigger "refresh" also)

### MeerK40t Version

current source

### MeerK40t Type

Git

### Your Operating System

Windows

kostex, 2024-01-18:
I've pulled the changes, but the issue is still there.. when disabling the primary grid and moving the object, it still snaps.. until zoom in/out
Thanks for your quick response anyhow :-)

jpirnay, 2024-01-18:
Can't confirm this behaviour, do you still have snap to elements active?

kostex, 2024-01-19:
Maybe I'm interpreting it all wrong..
I've got the snap to grid ON in preferences (snap to elements is OFF btw)
When I right click in the viewport and select/deselect "Show" Primary grid, the grid either is shown or only shown when moving a selected object.
Technically speaking this is expected behavior.. because the option is SHOW/HIDE grid.. not ENABLE/DISABLE grid
But, if I uncheck Show Primary Grid and zoom in/out a bit, the objects no longer snap either although the preference option snap to grid stays on.

So in my perfect world the show/hide primary grid just does that, and added right click options to enable/disable the grid/element snaps

Hope it's clear what happens.. 
Otherwise I can create a little video showing the behavior

Thanks!!
Roel

jpirnay, 2024-01-19:
You aren't interpreting it wrong, there was ambiguity in the consequences of the toggle grid logic, but as it leads to a definite behaviour after a zoom change it made sense to apply your interpretation. I tested it again after your feedback and it seems to behave consistently along these lines. So yes, a small video to display the issue would be helpful

kostex, 2024-01-19:
Here's a small clip showing the behavior I'm trying to explain.

https://youtu.be/DWCmFlndYcE

jpirnay, 2024-01-19:
Thank you for the video, it's not what I am experiencing - so are you really on the latest code base?
[image]

kostex, 2024-01-19:
Thanks for your video..
I saw that you're right clicking on the Ruler!! With lots more options!
I was not aware of that functionality.

I'm clicking in the viewport near the text!
Then the right click menu is different

jpirnay, 2024-01-19:
Now we're talking... Fixed, thanks for your patience

closed 2024-02-19 by jpirnay