meerk40t/meerk40t #2425: [Suggestion]: Add Raster/Engrave/Cut Operations to Material Editor

kostex, 2024-01-25:
### Summary Description

Either I don't know how or the feature is missing:
I'm creating a material database but cannot find a way to add an operation to a material.
If I want to add a Cut, Engrave or Raster operation to the list of operations, to my knowledge there is no way.

[image]

### Current functionality

Create a List of operations outside of the material editor and import them.

### Future functionality

Right click and add Raster, Engrave and or Cut operation in the material library editor

### MeerK40t Version

Latest commit

### MeerK40t Type

Git

### Your Operating System

Windows

jpirnay, 2024-01-25:
The way to allow full fledged editing of the material database entry is to load the set into the tree, edit it there with all the regular features and then to reapply the tree content to the material entry.

jpirnay, 2024-01-25:
Added some functionalities

closed 2024-02-19 by jpirnay