meerk40t/meerk40t #2437: [Bug]: Edit nodes: Nodes do not snap to grid

kostex, 2024-01-31:
### Summary Description

When snap is on, nodes do not snap to grid in edit mode.
[image]

### MeerK40t Version

Latest commit

### MeerK40t Type

Git

### Your Operating System

Linux

jpirnay, 2024-01-31:
Addressed in https://github.com/meerk40t/meerk40t/commit/f702dc4cd4e78cbde414e080c4cc2f5045dcdca6

kostex, 2024-02-05:
Something still is off.. I would presume that whenever a grid point has a green cross the node will snap when clicked, but that is not what happens.. zooming in and out feels like it changes the threshold too.
Adding curved points by holding the left mouse button and dragging, NEVER snaps to a grid point.
Hope I'm explaining it correctly.

Cheers!
[image]

jpirnay, 2024-02-05:
Vector tool seems to be the last pocke t of resistance - not understood why at the moment

Sophist-UK, 2024-02-05:
> the last pocket of resistance

It's those plucky Ukranian resistance fighters putting up a fight against the forces trying to exterminate them. 😄

closed 2024-02-19 by jpirnay