meerk40t/meerk40t #2456: From Version 0.9.2 Virus checker reports the file contains malware

KZNBro, 2024-02-09:
NOT A BUG IN THE SOFTWARE!

Hi, I'd like to report that from Version 0.9.2, downloading the meerK40t.exe is being rejected by my virus checker as the file reports the following malware:
The following is as transcribed from my AV notification screen=>

"Threat Blocked
We've blocked srsYR044.exe.part because it was infected with Win32:Malware-gen"

This is using AVG 

SW Version 23.12.3314 (build 23.12.8722.815)

Virus definitions Version: 240118-6

Number of definitions: 18,616,656

UI version: 1.0.737

Note that this warning is not apparent in Version 0.9.1 and is thrown out in ALL subsequent versions starting at 0.9.2 to current, 0.9.4 Beta 3 Bug Fixes and Adjustments. The person that has been compiling the files, the .exe at least, has malware on their machine. It would be much appreciated if that could be addressed as a matter of urgency.

This is a:

*   [] Bug
*   [] Suggestion for an enhancement

## SUMMARY

## DETAILS

## ERROR MESSAGE
```
<!-- Please copy and paste any error messages from e.g. a crash report file. -->
```

## YOUR ENVIRONMENT

*   MeerK40t version:
*   OS: 
*   Running from: <|-- e.g. Executable, package (PIP3/Pypi), downloaded source, git managed source

### SCREENSHOTS

### Zipped SVG File(s)

Sophist-UK, 2024-02-09:
@KZNBro Just to give a little more explanation, Meerk40t uses the same tool to package up Python code into Windows executables that many other projects use - and some of those projects are malware, and then all such projects get tarred with the same brush because they have the same signature because they use the same packaging tool.

There is, I believe, some sort of variable value you can set which changes the executable signature, so the devs have to experiment until they find one which doesn't trigger a false-positive virus alert and use that. And then, when an AV tool detects MK as a malware signature again, it doesn't mean that MK has a virus - it just means that some malware developer has started to use the same signature. Since the source code including the packaging code for the executable are open source and visible, you can still be confident that it doesn't include a virus.

TullioC, 2025-02-10:
Windows Defender: Trojan:Win32/Wacatac.B!ml

closed 2025-06-16 by jpirnay