meerk40t/meerk40t #2470: [0.9.4] Toolbar auto-sizing inconsistent

Sophist-UK, 2024-02-24:
[image]
Varying the height of the panel bar you get icons varying in size at different times rather than together.

jpirnay, 2024-02-27:
Actually the icon size is okay but the label position is awkward, addressed

Sophist-UK, 2024-02-27:
Jens - I don't know what change you made, but perhaps I should have been a bit more explicit about the issues:

1. Icons should always be top-justified and horizontally centred. 
2. Text should always be bottom-justified and horizontally centred.
3. Icon should be maximum size possible given the height of the text.
4. Down arrows should be bottom-justified **with the icon** (and not with the button.
5. One- or possibly also two-character words in the text should always be kept on the same line as the preceding word and not wrapped so e.g. Camera 0 should always be on one line.

IMO the algorithm should be:

1. Based on the number of button-groups and buttons, calculate the maximum button width (horizontal panel) or height (vertical panel) in order for all icons to fit. The maximum size of the other dimension is the fixed value of the panel.
2. Calculate the text size to fit the maximum-width of the button and (to prevent short text becoming very high) at most either a third of the maximum height (no wrapping) or half the maximum-height (wrapping) - calculate the height of this text inc. margins. See about about text wrapping.
3. Calculate the icon size to fit the space left after removing the vertical space for the text.
4. Find the maximum icon and text width and (if necessary) reduce the button sizes to fit this maximum.
5. Find the maximum icon and text height, and set a maximum button height based on that.
6. Place the graphic horizontally centred and vertically at top in the button space.
7. Place the text horizontally centred and vertically at bottom in the button space. To avoid too many switches of font size and improve visuals, set the text font size for each button to the minimum font size in the button group.
8. Please any down-arrow horizontally at right and vertically at the bottom of the space occupied **by the icon**. Currently the button sometimes overlays the text rather than the icon - and it seems to be a distance from the button top which is min(button-height, fixed value).

But I am not coding this, so please use or ignore these thoughts as you see fit.

closed 2024-10-04 by jpirnay