meerk40t/meerk40t #2487: [Suggestion]:

FelixEL, 2024-04-27:
### Summary Description

Some operations work better with air assist switched on (cutting) and some with off (engraving). @jpirnay  implemented a very good backbone of air assist with all the features needed for global air assist.

I started to work on a way to enable this feature for selected operations:  

[image]

This option will be also available form operation properties menu (double click on operation):
[image]

### Current functionality

Currently air assist can be set to on or off for all operations at once as part of configuration:
[image]

### Future functionality

Enabling air assist for selected operations - engraving - off, cutting - on etc. (user can do selection according to own needs).
Global air assist override (the one from configuration) - it makes sense to keep exit action (for example switch off air assist after all operations). Entry action related to air assist will be overridden with first operation specific air assist settings

### MeerK40t Version

86c8c45

### MeerK40t Type

Git

### Your Operating System

Windows

jpirnay, 2024-04-28:
Sounds legit, although you would need to have three states per laser operation: turn on - turn off - keep the existing state.
I will work on it.

FelixEL, 2024-04-28:
@jpirnay Sounds cool! I like idea of three state approach. Feel free to cherry-pick from my devbranch 

https://github.com/FelixEL/meerk40t/tree/airassist_grbl

I did some work (mostly GUI and callbacks). I was trying to figure out command injection for particular operation but couldn't find documentation how its done.

jpirnay, 2024-05-01:
Implemented and integrated in current main

closed 2024-05-01 by jpirnay