meerk40t/meerk40t #2488: [Bug]: circles as object show a smal line break

cminssie, 2024-04-28:
### Summary Description

Hi Thanks for the great software!
I noticed that circle object show a small break/stair in the line.
by circle object I mean a circle drawn with the **merrk40t tools bar** or an inkscape circle drawn from the circle tool **without** converting it to a path.

I join an inkscape test file with 2 identical circles, but the right one is converted in a path (inside inkscape), the left one is an "object". The JPG show the small break in the circle perimeter.

Thanks.
[image]
[image]

### MeerK40t Version

0.9.4

### MeerK40t Type

Executable

### Your Operating System

Windows

jpirnay, 2024-05-01:
I don't think there's a deficiency in the treatment of the circle (or any other geometric element for that matter). It might be the latency of the laser stopping to fire while it's already accelerating towards the next shape. by default meerk40t uses the k40 feature of rapid movement between objects.
You can try to turn this feature off to see whether it makes a difference for you:
[image]

cminssie, 2024-05-01:
Thanks for looking at my case.I just tried the rapid movement settings, no change.Further tries show that the problem affects only the first circle whatever origin it is( object or path). Following circles are ok. I tried also with a rectangle instead of circle showing no problem.

Any idea ? Related to my hardware ?Thx

Yahoo Mail : Recherchez, organisez, maîtrisez 

grafik.png (view on web)