meerk40t/meerk40t #2490: [Bug]: M3-NANO fails with move commands

ezequielleonzybert, 2024-05-01:
### Summary Description

board: M3-NANO
Depending on the move value or move_absolut value, the command does nothing. It works with integers values but sometimes decimals makes it does nothing. This was talked on discord server and Tat told me how to solve this. I edited the code in "lihuiyu/driver.py" to change every "S1P/n" string to "S1PF/n". And now it's working, but I don't know if this is the correct solution neither how to do a pull request. So I leave this issue here. Thanks.

### MeerK40t Version

9.4

### MeerK40t Type

Executable

### Your Operating System

Windows

jpirnay, 2024-05-02:
Addressed in PR 2491

Sophist-UK, 2024-05-02:
Reposting the PR #2491 so it is linked.

closed 2024-08-02 by jpirnay