meerk40t/meerk40t #2510: Simulation Window issue on Linux Mint

Alb673, 2024-06-20:
### Discussed in https://github.com/meerk40t/meerk40t/discussions/2505

<div type='discussions-op-text'>

<sup>Originally posted by **Alb673** June 12, 2024</sup>
Hi peeps, I have done a quick search, but haven't found something similar to my issue.

I recently downloaded the app image Linux v0.9.4020
I have noticed yesterday that while clicking simulation, it opens with a blank window. I have attached a video of the issue.
[Simulation -Issue-2024-06-11T210129+0100.webm](https://github.com/meerk40t/meerk40t/assets/23157880/07429a06-1f17-48be-91ab-b7274c983c34)

Cheers </div>

Sophist-UK, 2024-06-20:
There is not a paid development team for Meerk40t so you will need to be patient.

jpirnay, 2024-06-20:
I had a phenomenon like this some months ago and addressed it (and it is still fixed for me with Mk 0.9.4.2, and Linux Mint under Oracle VirtualBox) so the usual questionnaire applies:
Which system, which window manager, which mk version etc.
```Systeminfo:
System:
  Kernel: 5.15.0-112-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
```
Meerk40t Help-About
```
Python: 3.10.12, Present 
wxPython: 4.2.1 gtk3 (phoenix) wxWidgets 3.2.2.1, Present 
numpy: 1.26.1, Present 
pillow: 9.0.1, Present 
pypotrace: potracelib 1.16, Present (fast) 
ezdxf: 1.1.2, Present 
pyusb: 1.2.1, Present 
pyserial: 3.5, Present 
opencv: 4.8.1, Present 
barcode-plugin: 0.2.6, Present 
qrcode: ??, Present 
barcodes: 0.15.1, Present 
clipper: ??, Present 
numba: 0.59.0, Present 

```

jpirnay, 2024-06-20:
If your able to run from source you can try the latest code on github

Alb673, 2024-06-20:
> There is not a paid development team for Meerk40t so you will need to be patient.

Fair enough...I posted a question here last week. Wasn't even sure if someone saw it, and I got told on Makerforum, to do it as "issue"

Alb673, 2024-06-20:
@jpirnay Here are some info:
System:
  Kernel: 5.15.0-112-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy

Python: 3.6.8, Present 
wxPython: 4.1.1 gtk3 (phoenix) wxWidgets 3.1.5, Present 
numpy: 1.19.5, Present 
pillow: 8.3.2, Present 
potracer: ??, Missing 
ezdxf: 0.15.1, Present 
pyusb: 1.2.1, Present 
pyserial: 3.5, Present 
opencv: 4.5.3, Present 
barcode-plugin: ??, Missing 
clipper: ??, Missing 
numba: ??, Missing 

I can definitely see that I have some stuff missing, but I wouldn't know why or how to fix it....

> If your able to run from source you can try the latest code on github

I'm not sure how

jpirnay, 2024-06-20:
> I'm not sure how
https://github.com/meerk40t/meerk40t/wiki/Install:-Source

We were able to recreate it with wxPython < 4.1 and python < 3.12 - and seem to have hit a roadbloack as the underlying wxpython library seems to work incorrect. With the aforementioned combination PY 3.12 and WXP 4.2 we face no issues.
So my recommendation is clearly to go the stony way and install all the libraries - it's a one time effort but will allow you to follow much more closely the mk development.

jpirnay, 2024-06-20:
For the time being you can as well use the workaround you show in your video: maximize / refresh (right click and fit to bed should work too)

jpirnay, 2024-06-21:
Should be circumvented now even under wxpython 4.1 and before - will be part of next beta

jpirnay, 2024-08-02:
Part of 0.9.5.b4

closed 2024-08-02 by jpirnay