meerk40t/meerk40t #2530: [Bug]: CLI: Can't make simple text bold/italic

nmartont, 2024-07-16:
### Summary Description

Hi everyone,

I am trying to etch a dynamically generated [barcode](https://github.com/meerk40t/meerk40t-barcodes) and other misc info on a device. I cannot find a way to set text weight to bold via the CLI. Is there any way I can achieve this?

If not, is it possible to do this in other ways via CLI commands? It is a requirement that the GUI must not be used during the process. I am using the CLI batch file runner right now. Can I somehow add the bold text as an svg, for example? I cannot see any way to add an SVG next to the generated barcode to the project.

Thank you!

### MeerK40t Version

0.9.3001

### MeerK40t Type

PIP3/Pypi Package

### Your Operating System

Linux

jpirnay, 2024-07-18:
Yes, there is:
``property-set font_weight 700``
will set the text to bold. Values are:

-   200 = lighter
-   400 = normal
-   700 = bold
-   900 = bolder

(Maximum value is 1000, you can currently crash meerk40t if you apply values beyond that threshold)

jpirnay, 2024-07-18:
And yes you can put it to italic to:
``property-set font_style italic``
(``property-set font_style normal`` will reverse it)