meerk40t/meerk40t #2539: [Bug]: Unable to run MeerK40t on RPi 3b+

VanDavv, 2024-08-02:
### Summary Description

I'm trying to run MeerK40t on RPi3B+ so that I have remote access to my laser whenever I have a project ready. Unfortunately, I cannot start MeerK40t using [the latest release](https://github.com/meerk40t/meerk40t/releases/tag/0.9.4000), I downloaded the [MeerK40t_Pi64.tar](https://github.com/meerk40t/meerk40t/releases/download/0.9.4000/MeerK40t_Pi64.tar) file

### Additional Details

I've tried installing dependencies from here - https://github.com/meerk40t/meerk40t/wiki/Install:-Source#ubuntu-2204

I also tried to update glib to latest version

```
$ sudo apt-get install libglib2.0-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libglib2.0-dev is already the newest version (2.74.6-2+deb12u3).
libglib2.0-dev set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

I performed a fresh OS install, did `apt-get update` and `apt-get upgrade`, rebooted the device many times to make it work

### Crash logs

```shell
hive@raspberrypi:~/MeerK40t_Pi64 $ DISPLAY=:0 ./MeerK40t 
/usr/lib/aarch64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/tmp/_MEIvjivmq/libgnutls.so.30: version `GNUTLS_3_6_3' not found (required by /usr/lib/aarch64-linux-gnu/gio/modules/libgiognutls.so)
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiognutls.so
/tmp/_MEIvjivmq/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /lib/aarch64-linux-gnu/libproxy.so.1)
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiolibproxy.so
/usr/lib/aarch64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/aarch64-linux-gnu/gio/modules/libgiognomeproxy.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiognomeproxy.so
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-yes-antialias.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/48-spacing.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 4: unknown element "description"
Fontconfig warning: ignoring UTF-8: not a valid region tag

(MeerK40t:1722): Gtk-WARNING **: 14:29:02.711: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/PiXflat/16x16/mimetypes/image.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted
```

### MeerK40t Version

0.9.4000

### MeerK40t Type

Executable

### Your Operating System

Raspberry Pi

joerlane, 2024-08-02:
I just uploaded the latest beta versions, could you try 0.9.4050 and see if that makes any difference?

What operating system are you running on the Raspberry Pi? 

How long did you wait for it to launch? It takes an unusually long time to load the GUI on my Pi 3B+ with 64 bit Ubuntu installed; a number of minutes actually.

You shouldn't need to worry about those dependencies unless you are trying to run MeerK40t from source. The GLIBC version inside the frozen binary build is 2.27; so you shouldn't have any problem if you system has something newer. You could double check with the command `ldd --version ldd`.

VanDavv, 2024-08-03:
Here's my OS version:

```
 $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

I've waited less than a minute before a crash, not that long I'd say.

I downloaded the latest release and the same issue is still here:

```
$ DISPLAY=:0 ./MeerK40t 
/usr/lib/aarch64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/tmp/_MEI2ShuuC/libgnutls.so.30: version `GNUTLS_3_6_3' not found (required by /usr/lib/aarch64-linux-gnu/gio/modules/libgiognutls.so)
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiognutls.so
/tmp/_MEI2ShuuC/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /lib/aarch64-linux-gnu/libproxy.so.1)
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiolibproxy.so
/usr/lib/aarch64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/aarch64-linux-gnu/gio/modules/libgiognomeproxy.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiognomeproxy.so
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-yes-antialias.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/48-spacing.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/70-no-bitmaps.conf", line 4: unknown element "description"

(MeerK40t:39387): Gtk-WARNING **: 15:32:41.597: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/PiXflat/16x16/mimetypes/image.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted
```

I downloaded the RPI_64 tgz from [here](https://github.com/meerk40t/meerk40t/releases/tag/0.9.4050)

joerlane, 2024-08-03:
Interesting. I tested it on my build box (which is 64 bit ubuntu running on a Pi 3B+) and it works fine.

Looks like I'm going to need to grab some microSD cards and install the 64 bit Debian OS versions and see what's going on there. Those did not exist when I built the first 64 bit MeerK40t Pi build system. 

Not sure exactly how long it will take. My downloads still have some hours; and if I have to build wxPython from source, that could take a significant amount of time. Not sure what is available these days. 

I will report back when I know more about the issue.

Thanks for the bug report; much appreciated.

VanDavv, 2024-08-05:
Thanks a lot @joerlane! Is there anything I could do to help you in resolving this issue?

I can even share my RPi with you, it's a side project of mine and I just wanted to see if I can make my new laser work with this package

I also have RPi 4 with me, I can give it a try there or try different OS on my RPI 3b+ if you recommend

joerlane, 2024-08-05:
I was able to recreate the error you found with the same OS you have installed. I'm currently trying to build wxPython from source. That can be time consuming; but would allow me to build a binary that would certainly work on your current OS install. Then I could work out the details of the failure later.

Another thing I thought about was trying to build a modern wxPython wheel on my current 64 bit MeerK40t build box (which might be necessary if this wxPython build fails).

One final note; it appears the current 64 bit binary does actually function without the GUI. If you were planning to run the Pi headless, and address MeerK40t over TCP anyway from another computer on the network, then that could likely also work in this situation (at least short term while I'm working out a formal solution). 

It has been taking me 4 hours to write the OS to microSD cards, so I'll need a bit more time to verify myself. But I assume going backwards one OS version to Bullseye, and/or trying a 32 bit OS (with 32 bit Pi MeerK40t) might work for you short term as well. 

I'll let you know when I have some further progress on this; I am actively working on it.

VanDavv, 2024-08-05:
I really appreciate your answer @joerlane! Even though I'm not in a hurry and just wanted to help, your answer blew me off in both a level of usefulness and active ways I can still use the MeerK40t while you're working on a fix - chapeau bas! 

I'll gladly wait as long as it takes and will try the TCP mode in the meantime, thanks a lot!

joerlane, 2024-08-06:
I got a wheel. I'm just taking my time figuring this out. I'll post a new 64 bit binary for you soon. Just need to build it and test it out before upload. 

I'm still celebrating. This was faster than I imagined.

joerlane, 2024-08-06:
I got Meerk40t running in the GUI on that 64 bit Debian OS. Just need to run through the paces for the binary build.

joerlane, 2024-08-06:
Here. Try this on for size. It's not full featured, but it does run on the OS you asked about. I can do more as needed; but I really wanted you to see it working.

MeerK40t_Pi64Deb.tar is now in the latest release 0.9.4050

https://github.com/meerk40t/meerk40t/releases/download/0.9.4050/MeerK40t_Pi64Deb.tar

VanDavv, 2024-08-06:
Awesome, I can confirm it works! Astonishing work man, kudos!

[image]

I can close the issue now if you'd like

joerlane, 2024-08-07:
No need to close it just yet. There are a few extra plug-ins I need to add in to the binary, and some other things that are performance enhancements. That binary contains all the required components, but there are a few other things that can be added. I'll let you know when I get that finished. 

I also want to circle back and figure out why the original 64 bit binary is not working on the latest Debian 64 bit release, and that will also take some further time. I would like to update this bug report with those results before closing.

Ultimately I would like to fix the current 64 bit release so it functions everywhere, and/or move forward with something more compatible. That 64 bit Pi build box I was using is very old, from about the first moment a 64 bit OS became available for the Pi. There is likely some catch up work for me to do with that machine.

I really appreciate the bug report. I would not have noticed otherwise, given my current build machine runs the original MeerK40t 64 bit release (that it generates) just fine.

closed 2024-10-04 by jpirnay