meerk40t/meerk40t #2559: [Bug]: Movement buttons disabled after login

GazButane, 2024-08-21:
### Summary Description

After connecting the laser via USB and clicking **connect** in the **controller** window, the buttons to move the laser turn gray when hovering. It is impossible to move the laser head unless you restart the software (once restarted, it works perfectly).

### Additional Details

1. My laser engraver is an AtomStack A10 Pro
2. My linux distribution is based on debian

### Crash logs

```shell
/
```

### MeerK40t Version

v0.9.3010

### MeerK40t Type

Executable

### Your Operating System

Linux

jpirnay, 2024-08-21:
Admittedly this is Windows and the latest version:
[image]
But can you please verify your finding with the latest version - that would be 0.9.b5 - but as we are right now immediately before the launch of a new beta, you could try the Linux version already: https://github.com/meerk40t/meerk40t/releases/download/untagged-f422dc9cfb309fcb7cba/MeerK40t-Linux-Ubuntu-22.04
In any case, did you see a successful response from the laser in your connection window?

GazButane, 2024-08-22:
Hello, thank you for your answer
I did some tests with the latest version and the bug I encountered previously seems not to have recurred! Following these tests I still noticed that when I plug the laser into the computer AFTER starting Meerkat, the USB port is not detected automatically and it is necessary to select it manually in the settings. no problem however when I plug the laser BEFORE opening Meerkat

closed 2024-10-07 by jpirnay