meerk40t/meerk40t #2564: [Bug]: Documentation of reading scale & correction form cor file

Eszartek, 2024-09-01:
### Summary Description

I'm using a JCZ Fiber laser. I can see in the code where Meerk40t will read in a cor file for scale and lens correction, but I dont see any documentation for the format of that file, other than what I can read in lmc_controller.py. I currently scale my display by fudging the Width parameter in the Balor device setup page since there are no scaling options for this type of  device that I can find.

### Additional Details

From other Meerk40t code I've read,  I see the lens correction gui is incomplete/work in progress, but if scale and lens correction is read and used in Meerk40t, then if I had a sample of a valid cor file, I can work to create one for my machine. Does anyone have one they can share?

### MeerK40t Version

0.9.4060

### MeerK40t Type

Executable

### Your Operating System

Linux

joerlane, 2024-09-01:
Some is known about the format of the .cor file, but the ability to create them from scratch inside MeerK40t was never completed. 

A .cor file can be created by one of the CorFile.exe utilities included with the EZCad software on a windows PC. You will need to create one for each lens, since every lens has a unique distortion. The utility will draw a grid on a test material, and then you will need to measure each span, and enter those measurements into the software. That will generate a .cor file which can be copied and used inside MeerK40t.

Eszartek, 2024-09-01:
Thanks Joe! I'll give it a try, I have 2 lenses to work with. I tried using the Markcfg7 file that came with my Laser, but it messed up the engraving area and made the galvos chatter. I didn't see any header in my file like "LMC1COR_1.0" so I thought it may be a new file format, since the code is trying to load a 65x65 table of numbers, but there is no table data in my cor file either.

joerlane, 2024-09-01:
I'm currently trying to see if I can get any of the CorFile.exe or variants to run in WINE which should work on Linux and Mac, assuming MeerK40t sends the raw grid to the laser. I have not succeeded in getting WINE to run the .exe files yet today, though I'm reasonably sure I had managed to get it working in the past. I will keep working on that.

I do know that we have a capture of the grid it draws already, and MeerK40t could be made to do that part without much trouble. 

You only need to enter 16 measurements, and the CorFile.exe software converts those into the full 65x65 table of numbers which is the majority of what's inside the .cor file.

Eszartek, 2024-09-02:
I have a Windows vm to run the exe in, so thats not an issue. I used the Meerk40t console the other day to pull up the correction app, and it did have a button to send the grid I think. Ill try that again. If not, I'll do the Windows bit to get the cor file created, but then Ill work on writing a cross platform CorFile.exe. If I can do that in the one of the languages I program in daily (c or tcl), then I may try doing a python version as my first python app.

Eszartek, 2024-09-02:
The existing widget_corfile does burn the correct pattern and has the gui for entering the measurements for the pattern, but theres just no "write corfile" function implemented yet. So, once I have an understanding of the cor file table generation, that Meerk40t function can be finished in short order.

joerlane, 2024-09-02:
You'll need to install matplotlib and dependencies for this; but this is a script used to visualize the contents and meaning of a completed .cor file. Just drop the .txt off the end (github wouldn't let me upload a .py directly).

[cor_file_viewer10.py.txt](https://github.com/user-attachments/files/16831307/cor_file_viewer10.py.txt)

jpirnay, 2024-09-03:
Currently working on an implementation of the write routine.

Eszartek, 2024-09-03:
Thank you joerlane for that viewer. It will be interesting for me to get a feel for whats going on under the hood.

jpirnay: I'm glad to hear that this is on your todo list. I really like the way the current correction screen was designed, it is much more logical than the Ezcad interface. 

For the small items that I am engraving at the moment, correction isn't a big deal. One of my lenses is a 430mm with an engraving area of 300x300mm, and engraving something that large  will need correction. I look forward to help test your work in a future beta. I'll try generating a correction file in Windows in the mean time so I have something to compare with.