meerk40t/meerk40t #2581: [Bug] 0.9.5b6 Warning symbol shows when operations are deliberately disabled.

Sophist-UK, 2024-09-17:
This is usually a deliberate action... (and I don't want to see this)
[image]

But on the other hand elements not associated with an operation is often a mistake... (and I do want to see this)
[image]

jpirnay, 2024-09-17:
Feature request
[image]

jpirnay, 2024-09-18:
It always did: https://github.com/meerk40t/meerk40t/commits/main/meerk40t/gui/gui_mixins.py
But that's irrelevant as you can now set the cutoff threshold:
-Critical: anything that can harm your laser, ie bumping into rails
-Normal: anything that can spoil your burn, ie non-assigned elements
- Low: anything, that can spoil your burn, but you did it intentionally, ie disabled operations, hidden elements

```python
        self._concerns.clear()
        max_speed = getattr(self.context.device, "max_vector_speed", None)
        if has_ambitious_operations(max_speed, ("op cut", "op engrave")):
            self._concerns.append(
                (
                    _("- Vector operations are too fast.")
                    + "\n  "
                    + _("Could lead to erratic stepper behaviour and incomplete burns."),
                    CONCERN_CRITICAL
                )
            )
        max_speed = getattr(self.context.device, "max_raster_speed", None)
        if has_ambitious_operations(max_speed, ("op raster", "op image")):
            self._concerns.append(
                (
                    _("- Raster operations are too fast.")
                    + "\n  "
                    + _("Could lead to erratic stepper behaviour and incomplete burns."),
                    CONCERN_CRITICAL
                )
            )
        if has_objects_outside():
            self._concerns.append(
                (
                    _("- Elements are lying outside the burnable area.")
                    + "\n  "
                    + _("Could lead to the laserhead bumping into the rails."),
                    CONCERN_CRITICAL
                )
            )
        flag, info = has_close_to_edge_rasters()
        if flag:
            self._concerns.append(
                (
                    _("- Raster operations get very close to the edge.")
                    + "\n  "
                    + _("Could lead to the laserhead bumping into the rails.")
                    + "\n  "
                    + info,
                    CONCERN_NORMAL
                )
            )

        non_assigned, non_burn = self.context.elements.have_unburnable_elements()
        if non_assigned:
            self._concerns.append(
                (
                    _("- Elements aren't assigned to an operation and will not be burnt"),
                    CONCERN_NORMAL
                )
            )
        if non_burn:
            self._concerns.append(
                (
                    _(
                        "- Some operations containing elements aren't active, so some elements will not be burnt"
                    ),
                    CONCERN_LOW
                )
            )

        non_visible, info = has_hidden_elements()
        if non_visible:
            self._concerns.append(
                (
                    _("- Elements are hidden and will not be burnt") + f" ({info})\n",
                    CONCERN_LOW
                )
            )

```

Sophist-UK, 2024-10-01:
I just had unclassified elements **and** a disabled op and because I was expecting the warning for the disabled op, I did a burn with missing elements!!!!

jpirnay, 2024-10-01:
Warninglevel needs to be critical+normal

Sophist-UK, 2024-10-01:
I don't understand that. I guess I need some explanation as to what each means, and what selecting them means because it is completely unclear at the moment.

Sophist-UK, 2024-10-03:
[image]

I thought this was working yesterday, but today not.

I changed it bac to critical and reclassified. Then back to Normal+Critical and reclassified. Then took the screenshot.

Sophist-UK, 2024-10-03:
Warning symbol needs to be recalculated (event triggered) when the preference dropdown is changed.

jpirnay, 2024-10-03:
Thank you for your comment, this has already been implemented in PR #2613

Sophist-UK, 2024-10-03:
Can we please improve the tooltip in Preferences / GUI to explain what Low, Normal & Critical actually mean?

closed 2024-10-04 by jpirnay

Sophist-UK, 2024-10-15:
Still wrong in 0.9.4080:
[image]

And no definitions of what `Critical`/`Normal`/`Low` mean.:
[image]

Sophist-UK, 2024-10-15:
Code here does not appear to take account of the UI settings: https://github.com/meerk40t/meerk40t/blob/061c80507c79e6ef2919ee1bc7de597cdcc6e458/meerk40t/gui/wxmtree.py#L1079-L1093

PR #2645 (2024-10-15): Add info tooltips
- Recognize setting for tree warnsymbol
- Add explanations to preference option

[image]
Resolves #2581

## Summary by Sourcery

Introduce tooltips to explain warning severity levels in the GUI and enhance the logic for displaying warning messages based on these levels.

New Features:
- Add tooltips to provide explanations for different warning severity levels in the GUI.

Enhancements:
- Improve the logic for displaying warning messages based on user-defined severity levels.