meerk40t/meerk40t #2608: 0.9.5b7: New icon on windows taskbar needs white background for dark themes

Sophist-UK, 2024-10-01:
[image]

Sophist-UK, 2024-10-15:
Windows 10 dark theme.

MK pinned to taskbar but *not* opened i.e. shortcut icon:
[image]

MK opened i.e. executable icon:
[image]

The contrast on both of these is not great. Unopened it is white on white. Opened it is black/white on grey.

In both cases, my recommendation is to:

* Shortcut icon: Make the Meerkat Black with red eyes just like the execution icon
* Both icons: Make the circle red - and perhaps a tiny bit thicker - and fill with white background inside the circle to give contrast with meerkat and laser-eyes and circle.
* Both icons: Background outside the circle either transparent or black or dark grey.

closed 2024-11-08 by jpirnay

Sophist-UK, 2024-11-10:
Sorry - still not great in 0.9.5.3:
1. Icon when running too small and so it is VERY indistinct - it really does not do the brand recognition thing that it needs to do. I provided the Discord icon for comparison.
[image]
2. The shortcut icon is a better size, but the MK body is white rather than black.
[image]

jpirnay, 2024-11-10:
Look at the one on github and validate your findings

Sophist-UK, 2024-11-10:
Which one on Github? If you mean you want me to run from source, then sorry - but after I left the development community due to being abused I deleted my development environment.

But the above graphics are on the absolute latest executable from only last week.